qmk/qmk_firmware #19882: [Bug] Python 3.10 Requirement

spencerthayer, 2023-02-18:
### Describe the Bug

I currently have the latest Python installed on Mac OS and QMK is looking for an older version.
```
/usr/local/bin/qmk: bad interpreter: /usr/local/Cellar/qmk/1.1.1/libexec/bin/python3.10: no such file or directory
```

Installing 3.10 with `brew install python@3.10` resolves this issue but the idea of running multiple versions of Python is unacceptable. It's not like QMK is using Python 2 or some legacy code, it's simply not allowing for minor version updates.

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

tzarc, 2023-02-18:
As far as I can tell, `qmk` only currently depends on `python`, not `python@3.10`:
https://github.com/qmk/homebrew-qmk/blob/master/Formula/qmk.rb#L26

spencerthayer, 2023-02-18:
Which is all the more confusing why `brew install python@3.10` works.

tzarc, 2023-02-18:
I'd imagine it's due to when the package was built. Have flagged it internally, someone with a bit more expertise in this area might have a bit more info.

Perhaps just a republish is in order? Unsure.

ForsakenRei, 2023-02-23:
My two machines are running Python 3.9 and 3.10 but didn't have issue on both(Win and Arch though). 
I'm not a Mac user but this sounds like a Python issue rather than QMK issue. Probably check `which python` and then `ls -l` the path first?

android606, 2023-02-28:
I can confirm this issue. I run `qmk setup` and get:
`zsh: /usr/local/bin/qmk: bad interpreter: /usr/local/Cellar/qmk/1.1.1/libexec/bin/python3.10: no such file or directory`

 Installing on Mac OS with `brew install qmk/qmk/qmk` creates a symbolic link `/usr/local/Cellar/qmk/1.1.1/libexec/bin/python3.10` that points to
`/usr/local/opt/python@3.10/Frameworks/Python.framework/Versions/3.10/bin/python3.10`

I don't have python 3.10 installed, so that symlink is dead. Homebrew has dependency for 'python' but installed python 3.11, and the symlink doesn't point to it. This would appear to be an issue with qmk cask.

fauxpark, 2023-03-01:
I've updated the formula, please try it now.

spencerthayer, 2023-03-02:
Working!

closed 2023-03-02 by spencerthayer