qmk/qmk_firmware #19899: [Feature Request] Native support for Fast Fourier Transform (FFT) in plain C?

myst729, 2023-02-21:
### Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [X] New behavior

### Description

Recently I'm trying to implement sound spectrum visualizer with a microphone module following [this post](https://www.hackster.io/javagoza/sound-spectrum-visualizer-with-arduino-nano-33-ble-3c6261). And I've successfully made it work on a Pi Pico (supported by QMK now).

So I would think of making it work with a QMK keyboard (maybe with RGB lights as well). The core functionality relies on an FFT library to transform sound captured by the MAX9814 module into a series of analog input. However, the lib is written in CPP that doesn't seem to work with C now. And it's obviously beyond my skill sets to make necessary modifications.

So uh... is this idea interesting enough to get into the feature backlog and take effort? If so, any ideas to either make the lib work, or rewrite one from scratch in C, or does such a plain C lib already exist?

myst729, 2023-02-21:
I tried to port it to C, along with an example, https://github.com/myst729/qmk_firmware/tree/master/keyboards/dg/rgb36/fft

The original files are here: [lib](https://github.com/kosme/arduinoFFT/tree/master/src), [example](https://github.com/kosme/arduinoFFT/blob/master/Examples/FFT_01/FFT_01.ino).

Reviews? 🤣

myst729, 2023-02-22:
Well... I spent some time to port one https://github.com/myst729/fft , and it works fine on my Pi Pico via Arduino IDE. I'll try to add it to a QMK keyboard, let's see what happens 🤣

myst729, 2023-02-22:
Tried with an [Adafruit MAX9814 microphone module](https://www.adafruit.com/product/1713), and a homemade 6x6 WS2812 RGB breakout board.

https://user-images.githubusercontent.com/519733/220703979-9c06577a-deb6-4a23-9fd4-b377d85c1928.mp4

It works, but not as the same as running Arduino application. I guess the QMK main loop is too slow in this scenario (the sound sampling frequency may reach 30kHz or higher)?

myst729, 2023-02-23:
Even worse on an OLED display. Seems not a good idea of QMK feature.

getreuer, 2023-02-27:
Cool project! I don't know how many QMK users there are that want spectrum analyzers or have other needs for FFTs. Realistically, probably not enough that QMK wants to be on the hook for supporting one... it does get easily demanding to make a good FFT implementation that is both fast and supports the range of options that users may expect.

FWIW, here are a couple FFT implementations in pure C:

* [KissFFT](https://github.com/mborgerding/kissfft) is nice FFT library in pure C: it's not the fastest implementation, but it supports a good range of options (non-power-of-2 sizes, real-to-complex, fixed-point data), and the code is relatively simple.

* Or for even simpler, I have a [DSP library including a small FFT implementation](https://github.com/google/audio-to-tactile/blob/main/src/dsp) in pure C, intended for embedded systems with native float32 ops. It does transform sizes 4, 16, 64, 256, and 1024 without heap allocation.

myst729, 2023-02-28:
@getreuer Thanks for great information, I'll take a look on weekends.

However, the Arduino application (my port version of FFT) runs well, so I guess FFT lib may not be the bottle neck. The sound sampling rate is about 30KHz or higher, but I do this [in a matrix scan](https://github.com/myst729/qmk_firmware/blob/master/keyboards/dg/visualizer/keymaps/default/keymap.c#L57), which is usually slower that 1KHz. I guess this may be the reason it feels laggy? Is there a better choice?

getreuer, 2023-02-28:
Thanks for that! I missed that your spectrogram is running well outside of QMK. Nice work!

Like you say, typical audio sample rates sure are high, so yes, doing anything per sample at such a rate is demanding.

For reading in the audio from the MAX9814 mic, I see in section 4.9.2 of [the Pi Pico data sheet](https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf) that there is a SAR ADC peripheral with a "free-running sampling" mode. In this mode, you can set up the peripheral to sample at a fixed rate and store the sampled values in a buffer (ADC FIFO). This takes the burden of sampling off of the MCU. Set "DIV.INT" to 2999 to have it sample the mic at 16 kHz, and set the ADC FIFO and DMA for a buffer size of 256 samples. Then, you'll need the MCU to act once every 16 ms to process each buffer's worth of audio, which is much more manageable than acting per sample.

Another thought: QMK uses ChibiOS on ARM chips as the OS. It might help to use `chThdCreateStatic()` from [the threading API](https://www.chibios.org/dokuwiki/doku.php?id=chibios:documentation:books:rt:kernel_threading) to make a separate thread for running the audio handling loop.

Good luck!

myst729, 2023-04-01:
Close for (it seems) irrelevant to QMK.

closed 2023-04-01 by myst729