qmk/qmk_firmware #19906: Serial communication between split keyboard sides doesn't seem to be working

jeremiah-miller, 2023-02-22:
### Issue Description

I originally made an issue about this [here](https://github.com/josefadamcik/SofleKeyboard/issues/175), and it looks like the issue is either the elite c itself or the firmware. Does anyone here know of anything else to investigate, or any other settings to tweak that might fix the issue? 

Copying what I wrote in the other issue here:

> I have my keyboard all soldered and everything, however I'm having issues getting the two halves to communicate. Here are the relevant things in the config.h files:
> 
> ```
> #define USE_SERIAL
> #define EE_HANDS 
> #define FORCED_SYNC_THROTTLE_MS 100
> #define SPLIT_CONNECTION_CHECK_TIMEOUT 500
> #define SPLIT_TRANSPORT_MIRROR
> #define SPLIT_MODS_ENABLE
> #define SERIAL_DEBUG
> #define SELECT_SOFT_SERIAL_SPEED 1
> #define SOFT_SERIAL_PIN D2
> ```
> 
> I flash with `qmk flash -kb sofle/rev1 -km foureight84 -bl dfu-split-right` for the right side and `qmk flash -kb sofle/rev1 -km foureight84 -bl dfu-split-left` for the left, and whichever side is connected to the computer works as expected.
> 
> I've also tried wiring vcc->vcc, gnd->gnd, and D2->D2 directly, instead of using the trrs cable, but not even that worked. The other half gets power, which I know since all the leds turn on, but keys on the secondary half don't work

waffle87, 2023-02-22:
Try adding `#define SPLIT_USB_DETECT` to your config.h file, then recompiling and flashing both halves.

jeremiah-miller, 2023-02-22:
I tried that and it still didn't work

poonai, 2023-02-26:
Hey @jeremiah-miller , 

I've also had to same issue.  (my pro micros are soldered directly)

Today, I made it working by applying the below config to `config.h` file

```
#define SPLIT_USB_DETECT
#define SPLIT_USB_TIMEOUT_POLL 10
#define SPLIT_WATCHDOG_ENABLE
#define SPLIT_WATCHDOG_TIMEOUT 3000
```

I hope this will help you!!

jeremiah-miller, 2023-03-01:
hmm, that still didn't work for me :( I'm guessing it's probably a hardware issue, none of the things I've come across have worked

closed 2023-06-29 by github-actions[bot]