qmk/qmk_firmware #19921: No match for argument: libusb-devel on fedora

8i8, 2023-02-23:
### Describe the Bug

When I run ./util/qmk_install.sh I get the message :

No match for argument: libusb-devel

libusb-compat-0.1.x86_64 is installed 

And the following compiler error:

tar: Exiting with failure status due to previous errors
gcc  -O2 -Wall `libusb-config --cflags` -c usbcalls.c -o usbcalls.o
/bin/sh: line 1: libusb-config: command not found
In file included from usbcalls.c:19:
usb-libusb.c:28:10: fatal error: usb.h: No such file or directory
   28 | #include <usb.h>
      |          ^~~~~~~
compilation terminated.

### Keyboard Used

ergodox, using erdodox_ez

### Operating System

fedora 37

### qmk doctor Output

```text
Traceback (most recent call last):
  File "/home/iain/.local/bin/qmk", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/iain/.local/lib/python3.11/site-packages/qmk_cli/script_qmk.py", line 76, in main
    import qmk.cli  # noqa
    ^^^^^^^^^^^^^^
  File "/home/iain/Documents/system/ergodox/qmk_firmware/lib/python/qmk/cli/__init__.py", line 14, in <module>
    from . import config
  File "/home/iain/Documents/system/ergodox/qmk_firmware/lib/python/qmk/cli/config.py", line 62, in <module>
    @cli.subcommand("Read and write configuration settings.")
     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/iain/.local/lib/python3.11/site-packages/milc/milc.py", line 623, in subcommand_function
    return self.add_subcommand(handler, description, hidden=hidden, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/iain/.local/lib/python3.11/site-packages/milc/milc.py", line 604, in add_subcommand
    self.subcommands[name] = SubparserWrapper(self, name, self._subparsers.add_parser(name, **kwargs))
                                                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.11/argparse.py", line 1185, in add_parser
    raise ArgumentError(self, _('conflicting subparser: %s') % name)
argparse.ArgumentError: argument {config,clone,console,env,setup,c2json,chibios-confmigrate,clean,compile,config}: conflicting subparser: config
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

I have tried both the python qmk program, as well as running the install script from within the software git repo (my preferred and usual method), but to no avail.
Would appreciate some assistance in getting this working again, has been working for several years on Debian, have recently switch to using fedora, this is the first time that I attempt to flash my keyboard since changing.

tzarc, 2023-02-23:
Can I confirm that your copy of `util/install/fedora.sh` matches the [latest](https://github.com/qmk/qmk_firmware/blob/master/util/install/fedora.sh)? This should already be fixed.

8i8, 2023-04-08:
Thank you for the heads up, I've re-read the docs and am now compiling from my own forked repo, I'd not paid enough attention to the requirements after having switched from ubuntu to fedora; Thank you for the heads up, all is working fine now.

closed 2023-04-08 by 8i8