qmk/qmk_firmware #19922: Analog support for STM32G431

HapiCZ, 2023-02-23:
### Feature Request Type

- [ ] Core functionality
- [X] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

Would it be possible to add support for STM32G431 to analog.c? New keyboards from Deskeys use this chip. It would be possible to install QMK firmware on these keyboards.

Cipulot, 2023-02-23:
Following this issue, addition of the ADC support for the brother of the G431, the G474, would be nice. In practice the interesting feature is the internal redirect of ADC outputs to the opamps without utilizing GPIO pins. 
First step in the line is for sure the ADC addition and then that can be implemented too. Some work has been shown in the QMK discord a while back.

HapiCZ, 2023-03-01:
I'm trying to figure it out. Although I am a programmer, this is completely new to me. It seems to me that the problem with the STM32G431 and STM32G441 chips is already in ChibiOS itself.

STM32_FLASH_NUMBER_OF_BANKS is set to 2 in the [stm32_registry.h](https://github.com/qmk/ChibiOS/blob/0062927e3058a8b5ef587234bbd98d42fb4e595e/os/hal/ports/STM32/STM32G4xx/stm32_registry.h#L608). According to the reference manual, these chips have only one bank.

In addition, the compilation includes [LLD/ADCv3/hal_adc_lld.h](https://github.com/qmk/ChibiOS/blob/0062927e3058a8b5ef587234bbd98d42fb4e595e/os/hal/ports/STM32/LLD/ADCv3/hal_adc_lld.h)

These two chips, however, only have ADC1 and ADC2.

I don't see how this can be solved without modifying ChibiOS. Can someone verify this? Maybe I'm wrong.

tzarc, 2023-03-01:
ADCv3 is the 3rd "version" of the ADC API within ChibiOS -- different chips have different APIs and the G4xx is set to use the v3 API. ADC1/ADC2 can thus be considered as peripheral instances that export the ADCv3 API.

For the dual-bank issue it you would likely need to lodge a bug report on the [ChibiOS forum](https://forum.chibios.org/viewforum.php?f=35) at this point.

Not sure what else is involved with the ADCs -- perhaps @Cipulot can link to the discussion directly?

Cipulot, 2023-03-01:
I was starting o look into the ADCs by myself but then in the last week, sigprof actually started doing some more explicit research and also posted considerations about default settings to implement.

Here's a [link](https://discord.com/channels/440868230475677696/867530715192885269/1079057392781762580) for the QMK discord server discussion. It's in the development channel and mixed inbetween other convos but it's all there.

In parallel, in this case mostly because of how I personally implemented the code, some considerations about the use of `ATOMIC_BLOCK_FORCEON` and interrupts in general were expressed by sigprof.

I think that the 2 could go forward in parallel, even tho that might take a while and will have to "take place" mostly in ChibiOS in the first part, to then move to QMK

sigprof, 2023-03-02:
Analog support for G4xx would probably work with the code that's currently in [my adc-stm32l4xx branch](https://github.com/qmk/qmk_firmware/compare/master...sigprof:qmk_firmware:adc-stm32l4xx) (it's just missing the chip-specific piece of code in `pinToMux()`). Beware that the branch is very likely to be rebased without any warning, and it also contains some unrelated changes in onekeys and the joystick code to make testing easier.

And, BTW, the EFL code for G4xx seems to have another problem — it defines `stm32_flash_get_size()`, which detects the actual flash size, but then does not use it anywhere, so the reported flash size might not match the real chip.

Cipulot, 2023-03-23:
@sigprof I just grabbed some hardware that uses the G431. I'll tinker with it a bit and test the implementation of your fork for the L4XX.

Will update along the way with developments.

closed 2023-07-23 by github-actions[bot]