qmk/qmk_firmware #19946: Pressing LT does not call process_record_user until TAPPING_TERM expires or another keypress happens

dropsonic, 2023-02-26:
### Describe the Bug

Dear QMK maintainers,

I observe behavior that seems entirely unexpected to me, so I would like to use some help.

I have an LT key in my layout (`LT(4,KC_TAB)`). I want to run some custom code each time this key is pressed, so I try to handle this keycode in `process_record_user`:

```c
bool process_record_user(uint16_t keycode, keyrecord_t *record) {
    switch (keycode) {
        case LT(4,KC_TAB):
            ...
            break;
    }

    return true;
}
```

The thing is that if I press this LT key, I don't receive the `LT(4,KC_TAB)` keycode until the `TAPPING_TERM` expires or another key is pressed. Do you know if this is expected? Can this event be handled immediately after pressing the LT key?

### Keyboard Used

ZSA Moonlander Mark I

### Link to product page (if applicable)

https://www.zsa.io/moonlander/

### Operating System

Windows 11

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

filterpaper, 2023-02-27:
Tap or hold decisions for these dual role keys are resolved by design, before they are passed to `process_record_user`. It will require core changes if you need to intercept before such decisions are made. 
Otherwise `record->tap.count` will inform you if it was tap or hold in `process_record_user`.

filterpaper, 2023-02-27:
Mod tap keys (including layer tap) keys intention are only resolved on "key up" before they are sent to `process_record_user` (and it is not a bug). I am not very clear about your scenario 1's intention, but [hold on other key press](https://docs.qmk.fm/#/tap_hold?id=hold-on-other-key-press) will activate `LT` hold immediately upon pressing any key, allowing the hold function to be activated earlier than 200ms.

dropsonic, 2023-02-27:
@filterpaper my intention is to handle the key presses as they physically happen so I can change the language in the OS in time, before any key presses happen on the symbols layer. Basically like this:

0 ms: pressed LT; send Alt+Shift immediately to change the language in the OS;
100 ms: pressed and released `@`; `@` can be `"` in Cyrillic layouts so I need to have English layout turned on in the OS;
100+ ms: released LT; send Alt+Shift again to switch the language back from English

With `process_records_user`, the first two steps won't have a delay in between, so the OS will receive Alt+Shift and `@` immediately one after another, won't be able to change the language that fast, and `"` will be typed instead of `@` (tested both on Windows and Linux).

filterpaper, 2023-02-27:
Your intention will not be solved with the in-built mod tap key. Perhaps you can implement the solution with a custom key code to handle your intentions directly on user space, including layer changes.

dropsonic, 2023-02-27:
@filterpaper thank you so much for your help!

Then I'll close the issue since it's the intended behavior.

closed 2023-02-27 by dropsonic