qmk/qmk_firmware #19989: [Bug] Keychron V1 constantly wakes computer from sleep

JamieKitson, 2023-03-02:
### Describe the Bug

Using the firmware that came with my Keychron V1 in December and the latest version 1.3 from the Keychron website the keyboard constantly wakes my computer from sleep. I have recorded a video demonstrating the issue here:

https://www.youtube.com/watch?v=gF9eFnA2qxc

The only interesting thing is that Event Viewer says that the reason for wake was a mouse input.

### Keyboard Used

keychron/v1

### Link to product page (if applicable)

https://www.keychron.uk/products/keychron-v1-qmk-via-custom-mechanical-keyboard

### Operating System

Windows 10

### Is AutoHotKey / Karabiner installed

- [X] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

sigprof, 2023-03-02:
Did you actually try using the firmware built from the upstream QMK sources (e.g., from https://github.com/Xelus22/QMK-VIA-Hex/tree/master/keychron, if you want to have VIA support, or from config.qmk.fm)?  Does that firmware have the described bug?

Also you can try toggling the MacOS/Windows mode switch on the back to the other position and then repeating the test.  If the problem won't appear with the other switch position, then the stock Keychron firmware has a bug related to handling that switch.  However, the firmware built from the upstream QMK sources should not have that problem (it has the `MATRIX_MASKED` configuration that prevents the mode switch from sending the resume signal).

JamieKitson, 2023-03-02:
The bug is still reproduceable both with the firmware from the link you supplied and in mac mode.

skagon, 2023-05-22:
I have the same problem. Weird thing is, the computer (Windows PC) goes to sleep only to be immediately awakened upon final switch-off. It's like, the moment the connection is severed, the keyboard sends some signal to wake it up again.
Note: keyboards/keypads are connected through a powered USB hub, even though it's sometimes happening with unpowered hub.