qmk/qmk_firmware #20014: [Bug] SouthPaw 65 keyboard incorrectly configured

elegantalchemist, 2023-03-04:
### Describe the Bug

When using QMK configurator, the default layouts for the Southpaw65 (default, ANSI and ISO) are incorrect,

Selecting to compile one of the defaults gives a hex file, which when flashed has all the keys shifted 16 steps to the right. I suspect there is a bug/issue in the column/pin configuration.

### Keyboard Used

southpaw_65

### qmk doctor Output

```text
n/a
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

sigprof, 2023-03-04:
You mean the `switchplate/southpaw_65` board?

Looks like the code for that board had been broken by the changes in #16277 (maybe wrongly copied from `xiudi/xd96`).

elegantalchemist, 2023-03-04:
Yes that's right, that board

Something like that would make sense. The rows are fine, just that all columns are offset by 16 keys to the right. It's consistent across all rows, suggesting a config error

elegantalchemist, 2023-03-05:
Close as completed - pending pull request #20015  solves the issue.

PR #20015 (2023-03-05): Unbreak switchplate/southpaw_65 after #16277
## Description

Apparently the changes done in #16277 swapped the values read from the PCA9555 expanders by mistake, which resulted in mixed up matrix columns. Fix the chip and register addresses to match the original code.

The broken change is [this](https://github.com/qmk/qmk_firmware/pull/16277/files#diff-921203cf4ee047ea3650fecd282a5709f92b83c88c0cbfbf3542f7f4fb60a63b); looks like the sequence of `pca9555_readPins()` calls was mistakenly copied from `xiudi/xd96`.

@elegantalchemist, are you able to compile the code locally to test these changes [... cut]
closed 2023-03-05 by elegantalchemist