qmk/qmk_firmware #20022: New massage with 0.20 compilation that was not present with the 0.19

AndroSteph, 2023-03-05:
### Issue Description

Hello,

I recompiled my firmware for my Keychron Q6 with the new files put here "..qmk_firmware\keyboards\keychron\q6" and I have this new message that appears (compilation with "default, keychron and VIA)

Is it a problem (I don't understand the meaning of this message) but the compilation finishes without error and I flashed the new firmware without problem in my keyboard.

So in the end no problem, I just report this difference between version 0.19 and 0.20.

Thanks for all your work :-)

AndroSteph, 2023-03-05:
[image]

keyboard-magpie, 2023-03-05:
It's a warning, not an error.

closed 2023-03-05 by keyboard-magpie

tzarc, 2023-03-05:
See [here](https://docs.qmk.fm/#/ChangeLog/20230226?id=i-m-t-i).

AndroSteph, 2023-03-05:
Thank you very much for your answers ;-)