qmk/qmk_firmware #20024: Errors when trying to compile a keymap using the leader key functionality

DerWahreMatheMann, 2023-03-05:
### Issue Description

Hi, I'm trying to get the leader key functionality working but when compiling, I get several errors. The layout where I'm trying to use the leader key looks like this:

```
[_COLEMAKDH] = LAYOUT_planck_grid(
    KC_ESC,  KC_Q,    KC_W,    KC_F,    KC_P,    KC_B,    KC_J,    KC_L,    KC_U,    KC_Y,    KC_QUOT, KC_ENT,
    KC_TAB,  KC_A,    KC_R,    KC_S,    KC_T,    KC_G,    KC_M,    KC_N,    KC_E,    KC_I,    KC_O,    KC_BSPC,
    KC_LSFT, KC_X,    KC_C,    KC_D,    KC_V,    KC_Z,    KC_K,    KC_H,    KC_COMM, KC_DOT,  KC_DLR,  KC_RSFT,
    KC_LGUI, KC_LCTL, KC_LALT, KC_LEAD, NUMS,    KC_SPC,  KC_SPC,  SYMBS,   KC_LEFT, KC_DOWN, KC_UP,   KC_RGHT
)
```

After my keymap definitions I have the basic `matrix_scan_user` function for the leader key without any extra functionality (I removed any I had already added to see whether it would fix the issue):

```
LEADER_EXTERNS();

void matrix_scan_user(void) {
  LEADER_DICTIONARY() {
    leading = false;
    leader_end();
  }
}
```

When compiling i get this error:

```
Size before:
   text	   data	    bss	    dec	    hex	filename
      0	  54064	      0	  54064	   d330	planck_ez_base_test.bin

Compiling: quantum/keymap_introspection.c                                                          In file included from keyboards/planck/planck.h:22,
                 from ./.build/obj_planck_ez_base/src/default_keyboard.h:38,
                 from ./keyboards/planck/keymaps/test/keymap.c:1,
                 from quantum/keymap_introspection.c:5:
./keyboards/planck/keymaps/test/keymap.c:69:32: error: 'KC_LEAD' undeclared here (not in a function); did you mean 'KC_LPAD'?
   69 |     KC_LGUI, KC_LCTL, KC_LALT, KC_LEAD, NUMS,    KC_SPC,  KC_SPC,  SYMBS,   KC_LEFT, KC_DOWN, KC_UP,   KC_RGHT
      |                                ^~~~~~~
keyboards/planck/ez/ez.h:57:22: note: in definition of macro 'LAYOUT_ortho_4x12'
   57 |     { k37, k38, k39, k33, k34, k35 } \
      |                      ^~~
./keyboards/planck/keymaps/test/keymap.c:65:16: note: in expansion of macro 'LAYOUT_planck_grid'
   65 | [_COLEMAKDH] = LAYOUT_planck_grid(
      |                ^~~~~~~~~~~~~~~~~~
In file included from quantum/keymap_introspection.c:5:
./keyboards/planck/keymaps/test/keymap.c:109:1: error: data definition has no type or storage class [-Werror]
  109 | LEADER_EXTERNS();
      | ^~~~~~~~~~~~~~
./keyboards/planck/keymaps/test/keymap.c:109:1: error: type defaults to 'int' in declaration of 'LEADER_EXTERNS' [-Werror=implicit-int]
./keyboards/planck/keymaps/test/keymap.c:109:1: error: function declaration isn't a prototype [-Werror=strict-prototypes]
./keyboards/planck/keymaps/test/keymap.c: In function 'matrix_scan_user':
./keyboards/planck/keymaps/test/keymap.c:112:3: error: implicit declaration of function 'LEADER_DICTIONARY' [-Werror=implicit-function-declaration]
  112 |   LEADER_DICTIONARY() {
      |   ^~~~~~~~~~~~~~~~~
./keyboards/planck/keymaps/test/keymap.c:112:22: error: expected ';' before '{' token
  112 |   LEADER_DICTIONARY() {
      |                      ^~
      |                      ;
cc1: all warnings being treated as errors
 [ERRORS]
 | 
 | 
 | 
gmake[1]: *** [builddefs/common_rules.mk:360: .build/obj_planck_ez_base_test/quantum/keymap_introspection.o] Error 1
Make finished with errors
gmake: *** [Makefile:392: planck/ez/base:test] Error 1
```

I found another [post with a similar issue](https://github.com/qmk/qmk_firmware/issues/5224) where the fix was setting `LEADER_ENABLE = yes` in their rules.mk file but I had already done that before trying to compile, so that shouldn't be the problem.

tzarc, 2023-03-05:
See [here](https://docs.qmk.fm/#/ChangeLog/20230226?id=leader-key-rework).

DerWahreMatheMann, 2023-03-05:
Thanks a bunch! Don't know how I missed that but it fixed the issue.

closed 2023-03-05 by DerWahreMatheMann