qmk/qmk_firmware #20064: simple question about a different language than US

AndroSteph, 2023-03-08:
### Issue Description

Only one last question following my "How to configure keyboard other than US #20035" (I didn't want to reopen this ticket in order not to disturb the initial topic) https://github.com/qmk/qmk_firmware/issues/20035

I managed to modify the "keymap.c" file by adding #include "keymap_swiss_fr.h" and by changing some parameters like "KC_1" to "CH_1", etc.

I then managed to compile the whole thing without any error message.

My question is, what is the use of all this work? I didn't notice any changes that would have been important.

Indeed, I made a very simple macro and to make it work I had to replace "Y" with "Z" because of my language keyboard code which is not the same as the basic ANSI US.

I stupidly thought that by modifying the "keymap.c" file and once compiled and flashed in my keyboard I would not have this problem anymore. I always have to make sure that the text I want to see displayed (once my macro is run) to put an ANSI US code and not a swiss_fr.

For a simple letter, this is quite simple but very restrictive when it is a sequence of commands or text.

Is there a way to say in "VIA" (macro section) that if I want a "Y" (in my language) I can put a "Y" and not a "Z" (because of the inversion on my keyboard of these two letters, for example)

My English is not good, I hope my explanation will be understandable anyway.

Thank you for your advice and explanations.

fauxpark, 2023-03-08:
Please read the docs link I posted int he other issue. It explains what the point of these headers are.

AndroSteph, 2023-03-08:
My English is not very good (sorry), but I did not understand the real interest of doing all this work if it is only to "have a simple indication" it is in this idea that I really have difficulty to understand.

For the moment my conclusion is that it is useless. Except to have a more "accurate" "information" in the "keymap.c" file (so that something visual), but in everyday life it's totally useless since the keyboard uses the information of the OS (Windows for me) and VIA doesn't use anything, not even to display the right letters (of my language) on the keyboard representation.

I really have a hard time to understand this utility (I'm really sorry to take your time for a "simple" question for you but not for me)

fauxpark, 2023-03-08:
The keyboard doesn't know anything about the host or the language. All keyboards are exactly the same, the only difference is what's printed on the keycaps. These headers are similar, in that they don't contain separate keycodes to the `KC_` prefixed ones, they're just aliases so you can match what you've set your input language to in your OS. For example `KC_Q` (QWERTY) will do the exact same thing as `FR_A` (AZERTY).

AndroSteph, 2023-03-08:
Yes, this principle I understood it well. What I can't understand is why wasting time (in my tests yesterday to compile a "keymap.c" file with the right allias written in it) when this information is never used.

It's my OS that knows that if I use the Y key on my keyboard (the insciption is "Y" but the real function of this key in US ANIS is the "Z") my OS knows that I press the "Z" but that it should actually display a "Y" on my screen.

So why make useless changes in a "keymap.c" file, or more precisely why this possibility still exists to do it when it is useless... my question is about this, in what situation does it make sense?

fauxpark, 2023-03-08:
It's not useless, it's for readability.

closed 2023-03-08 by AndroSteph