qmk/qmk_firmware #20087: [Bug] AVR suspend/power down during sleep

jaygreco, 2023-03-11:
### Describe the Bug

I believe there is a long-running bug in AVR suspend during PC sleep. It manifests visibly with GPIOs: RGB goes haywire, the board isn't responsive after the PC sleeps until a power cycle, and on the Bit-C the built-in LED turns on. The common workaround is to define NO_SUSPEND_POWER_DOWN in the config.h, but this also keeps the RGB, backlight, etc enabled when the host is put to sleep. 

I've also seen a few cases where internal flash gets corrupted (at least on 32U4) and the board needed to be flashed again to bring it back to life. There have been a few attempts to work around and fix this issue, but despite that it pops back up from time to time. When I was originally digging into this, it seemed possible that it was related to clocks since almost all clocks are disabled during power-down, but I never got to a root cause. @drashna advised me to open this issue to brainstorm/track mitigations.

https://github.com/qmk/qmk_firmware/issues/19192
https://github.com/qmk/qmk_firmware/issues/14127
https://github.com/qmk/qmk_firmware/issues/6822
https://github.com/qmk/qmk_firmware/issues/6726
https://github.com/qmk/qmk_firmware/issues/5239

There's a separate class of issue related to sleep on M1 Macs -- this is different from that one. The two don't seem to be related.

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

skagon, 2023-05-29:
I think I'm affected by this on my ErgoDox keyboard (self-made). I'm not using any LEDs or lighting, I've taken the ErgoDox EZ code and stripped all irrelevant settings and pieces (relating to RGB LEDs and such), and it absolutely won't start up after a sleep/wake cycle. I have to unplug and reconnect it manually. I've also experienced flash corruption twice so far.
For the record, it's on a powered USB hub, along with a few other hand-wired keypads I've made, and only the ErgoDox exhibits this behaviour. [see #19871 for the others]

jaygreco, 2023-05-29:
Any chance of trying the fix in #20201? It cleared it up for me, although it hasn't seen any movement in the last few months. 

Another datapoint + a bump/comment (if it appears to help) would be helpful in getting it merged.

Kodehawa, 2024-01-08:
> I've also seen a few cases where internal flash gets corrupted (at least on 32U4) and the board needed to be flashed again to bring it back to life

I had this issue on my IK75v3 after sleep. That was really annoying.

jacropad, 2024-02-24:
I have a custom board with ATMega32U4 that's been having issues waking from sleep. It consistently fails when waking up, Windows displays the error "Unknown USB Device, device descriptor request failed".

`#define NO_SUSPEND_POWER_DOWN` seems to solve the problem, but I tried replacing that with `#undef wdt_vect` and the problem comes back unfortunately. But I'm content with using NO_SUSPEND_POWER_DOWN because I don't have any LEDs or anything to make it obvious that the board isn't asleep.