qmk/qmk_firmware #20089: [Bug] cannot compile using -e CONVERT_TO=kb2040

UwUham, 2023-03-11:
### Describe the Bug

i'm compiling my firmware for a dactyl manuform 4x5 right now, and i intended to use KB2040 microcontrollers. 
output of `qmk compile -kb handwired/dactyl_manuform/4x5 -km soul_layout -e CONVERT_TO=kb2040`:
```
Ψ Compiling keymap with make --jobs=1 CONVERT_TO=kb2040 handwired/dactyl_manuform/4x5:soul_layout

QMK Firmware 0.20.1
Making handwired/dactyl_manuform/4x5 with keymap soul_layout

arm-none-eabi-gcc.exe (GCC) 10.1.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: quantum/command.c                                                                       In file included from ./keyboards/handwired/dactyl_manuform/4x5/config.h:21,
                 from <command-line>:
quantum/config_common.h:19:9: note: '#pragma message: 'config_common.h' should no longer be included!'
   19 | #pragma message("'config_common.h' should no longer be included!")
      |         ^~~~~~~
 [OK]
Compiling: platforms/chibios/drivers/vendor/RP/RP2040/serial_vendor.c                              In file included from ./keyboards/handwired/dactyl_manuform/4x5/config.h:21,
                 from <command-line>:
quantum/config_common.h:19:9: note: '#pragma message: 'config_common.h' should no longer be included!'
   19 | #pragma message("'config_common.h' should no longer be included!")
      |         ^~~~~~~
In file included from platforms/chibios/drivers/vendor/RP/RP2040/serial_vendor.c:5:
platforms/chibios/drivers/vendor/RP/RP2040/serial_vendor.c: In function 'enter_rx_state':
platforms/chibios/drivers/serial_usart.h:15:33: error: 'A9' undeclared (first use in this function); did you mean 'A'?
   15 | #    define SERIAL_USART_TX_PIN A9
      |                                 ^~
platforms/chibios/drivers/vendor/RP/RP2040/serial_vendor.c:157:29: note: in expansion of macro 'SERIAL_USART_TX_PIN'
  157 |     gpio_set_drive_strength(SERIAL_USART_TX_PIN, GPIO_DRIVE_STRENGTH_2MA);
      |                             ^~~~~~~~~~~~~~~~~~~
platforms/chibios/drivers/serial_usart.h:15:33: note: each undeclared identifier is reported only once for each function it appears in
   15 | #    define SERIAL_USART_TX_PIN A9
      |                                 ^~
platforms/chibios/drivers/vendor/RP/RP2040/serial_vendor.c:157:29: note: in expansion of macro 'SERIAL_USART_TX_PIN'
  157 |     gpio_set_drive_strength(SERIAL_USART_TX_PIN, GPIO_DRIVE_STRENGTH_2MA);
      |                             ^~~~~~~~~~~~~~~~~~~
platforms/chibios/drivers/vendor/RP/RP2040/serial_vendor.c: In function 'leave_rx_state':
platforms/chibios/drivers/serial_usart.h:15:33: error: 'A9' undeclared (first use in this function); did you mean 'A'?
   15 | #    define SERIAL_USART_TX_PIN A9
      |                                 ^~
platforms/chibios/drivers/vendor/RP/RP2040/serial_vendor.c:170:59: note: in expansion of macro 'SERIAL_USART_TX_PIN'
  170 |     pio_sm_set_consecutive_pindirs(pio, tx_state_machine, SERIAL_USART_TX_PIN, 1U, true);
      |                                                           ^~~~~~~~~~~~~~~~~~~
platforms/chibios/drivers/vendor/RP/RP2040/serial_vendor.c: In function 'serial_transport_driver_master_init':
platforms/chibios/drivers/serial_usart.h:15:33: error: 'A9' undeclared (first use in this function); did you mean 'A'?
   15 | #    define SERIAL_USART_TX_PIN A9
      |                                 ^~
platforms/chibios/drivers/vendor/RP/RP2040/serial_vendor.c:437:20: note: in expansion of macro 'SERIAL_USART_TX_PIN'
  437 |     pin_t tx_pin = SERIAL_USART_TX_PIN;
      |                    ^~~~~~~~~~~~~~~~~~~
platforms/chibios/drivers/vendor/RP/RP2040/serial_vendor.c: In function 'serial_transport_driver_slave_init':
platforms/chibios/drivers/serial_usart.h:15:33: error: 'A9' undeclared (first use in this function); did you mean 'A'?
   15 | #    define SERIAL_USART_TX_PIN A9
      |                                 ^~
platforms/chibios/drivers/vendor/RP/RP2040/serial_vendor.c:456:20: note: in expansion of macro 'SERIAL_USART_TX_PIN'
  456 |     pin_t tx_pin = SERIAL_USART_TX_PIN;
      |                    ^~~~~~~~~~~~~~~~~~~
 [ERRORS]
 |
 |
 |
make[1]: *** [builddefs/common_rules.mk:360: .build/obj_handwired_dactyl_manuform_4x5_soul_layout/serial_vendor.o] Error 1
Make finished with errors
make: *** [Makefile:392: handwired/dactyl_manuform/4x5:soul_layout] Error 1
```
seems like something is wrong with the converter internally? if something i have is out of date, please link me to somewhere telling me where and how i can update them.

it compiles fine without -e CONVERT_TO=kb2040 if i #define USE_I2C, but serial instead of i2c is required for the 2040 from my understanding.

thanks!

### Keyboard Used

handwired/dactyl_manuform/4x5

### Operating System

Windows 10

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: C:/Users/Basil/qmk_firmware
Ψ Detected Windows 10 (10.0.19044).
Ψ QMK MSYS version: 1.7.2
Ψ Git branch: master
Ψ Repo version: 0.20.1
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest master: 2023-03-10 12:57:11 -0500 (df814dec7a) -- Caerdroia Keyboard Firmware (#20036)
Ψ - Latest upstream/master: 2023-03-10 21:33:52 -0800 (96b0baa230) -- [Keyboard] Add Spleeb default and custom firmware (#19481)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2023-03-10 12:57:11 -0500 (df814dec7a) -- Caerdroia Keyboard Firmware (#20036)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.1.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.0
Ψ Found dfu-programmer version 0.7.2
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-01-03 19:29:26 +0000 --  (0062927e3)
Ψ - lib/chibios-contrib: 2023-01-11 16:42:27 +0100 --  (a224be15)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [X] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

keyboard-magpie, 2023-03-11:
I suspect your submodules are out of date.

Run `make git-submodule`

UwUham, 2023-03-11:
```
[Basil@DESKTOP-N0PGUNM ~]$ make git-submodule
make: *** No rule to make target 'git-submodule'.  Stop.
```

keyboard-magpie, 2023-03-11:
The command needs to run from inside `/qmk_firmware` if you do not know how to navigate the command line you can type `qmk cd` then `make git-submodule`

UwUham, 2023-03-11:
ah, thanks, i'll try recomiling now

UwUham, 2023-03-11:
nope, it's the same error, any other ideas?

drashna, 2023-03-16:
config_common.h has been removed/deprecated.  It should no longer be included (and didn't really need to be in most cases).

Also, if `make git-submodule` isn't working from the qmk firmware folder, then you downloaded a zip instead of cloning the repo.

Also, I see serial defines listing A9 for the pin defines for serial, which is not a valid config for either avr nor rp2040.

myst729, 2023-05-04:
If you don't want to bother with QMK dev env, use https://config.qmk.fm/

closed 2023-11-01 by zvecr