qmk/qmk_firmware #20098: CAPS_WORD is a pain with kebab-case.

tecosaur, 2023-03-12:
### Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [X] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

I really like the idea of the caps word key, however, replacing `-` with `_` is quite a pain, for example when typing `CLEAR-ENTIRE-CACHE` I'd love to use caps word, but going back and manually replacing the underscores with `_` is quite a hassle.

While kebab-case is indeed more uncommon than snake_case, it's the standard in a number of programming languages (e.g. all lisp dialects), and so it would be good if caps word didn't only allow a single capitalisation scheme.

filterpaper, 2023-03-13:
There is an option to exclude keycodes from being "shifted" using `bool caps_word_press_user(uint16_t keycode)`. Have you tried [that feature](https://docs.qmk.fm/#/feature_caps_word?id=configure-which-keys-are-word-breaking)?

filterpaper, 2023-03-14:
Do mark this issue resolved if it works for you.

tecosaur, 2023-03-14:
It looks like it should, so I'll pre-emptively do so and come back if it doesn't.

closed 2023-03-14 by tecosaur