qmk/qmk_firmware #20135: [Bug] brew install returns "Formulae find in multiple taps"

orcinus, 2023-03-16:
### Describe the Bug

Running brew install qmk/qmk/qmk fails at:

```
Error: Formulae found in multiple taps: 
       * osx-cross/avr/avr-binutils
       * larsimmisch/avr/avr-binutils

Please use the fully-qualified name (e.g. osx-cross/avr/avr-binutils) to refer to the formula.
```

### Keyboard Used

N/A

### Link to product page (if applicable)

N/A

### Operating System

Mac OS Ventura 13.2.1

### qmk doctor Output

```text
N/A
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

Manually installing osx-cross/avr/avr-binutils does not seem to help.

orcinus, 2023-03-16:
Nvm.
Removed the offending larsimmisch/avr/avr-binutils tap, everything ok now.

Closing.

orcinus, 2023-03-16:
(That said, it might be a good idea to use the fully qualified path anyway.)

closed 2023-03-16 by orcinus

fauxpark, 2023-03-16:
We do.
https://github.com/qmk/homebrew-qmk/blob/master/Formula/qmk.rb#L25

orcinus, 2023-03-16:
Weird o_O