qmk/qmk_firmware #20137: External USB PHY support?

JunASAKA, 2023-03-16:
Hey guys, 
I am wondering if I can use a external USB PHY with STM32F to run USB High-Speed. How can I do with this firmware?

thanks.

tzarc, 2023-03-16:
Enable ULPI in ChibiOS, and change the USB driver to OTG2.

JunASAKA, 2023-03-16:
> Nope, it's a very specialised set of criteria, and based on your electrical circuit -- we don't document to that depth.
> 
> board.h will need:
> 
> ```
> #define BOARD_OTG2_USES_ULPI
> ```
> 
> mcuconf.h will need:
> 
> ```
> #define STM32_USB_USE_OTG1                  FALSE
> #define STM32_USB_USE_OTG2                  TRUE
> #define STM32_USE_USB_OTG2_HS               TRUE
> #define STM32_USE_USB_OTG2_ULPI             TRUE
> ```
> 
> config.h will need:
> 
> ```
> #define USB_DRIVER USBD2
> ```
> 
> You'll need to sort out configuration override files with the above info.

Got it. I will try 'em later.
Thanks!

JunASAKA, 2023-03-22:
And I am wondering how can I port this firmware to STM32H750 MCU which is supported by ChibiOS already. But I haven't found any documents regarding it. 
Anyone could tell me how to do?

Appreciate.

closed 2023-03-22 by JunASAKA