qmk/qmk_firmware #20152: [Bug] underglow modes on tiger80 (tiger lite) don't work anymore

danielpetre, 2023-03-16:
### Describe the Bug

My compilation of tiger80 qmk worked perfectly fine on 03.03.2023 but today (17.03.2023) the underglow rgb modes don't work anymore. It's just the plain static color light and cannot be change anymore. Please fix.

### Keyboard Used

kbdfans/tiger80

### Link to product page (if applicable)

https://kbdfans.com/collections/tiger-lite

### Operating System

Windows 11

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Other keyboard-related software installed

qmk_toolbox

dmeiners88, 2023-03-17:
## Describe the Bug
Same here. Keycodes `RGB_MOD` and `RGB_RMOD` have no effect. For reference, I use a custom keymap found here: https://github.com/dmeiners88/qmk_firmware/tree/tiger80-dmeiners88-keymap/keyboards/kbdfans/tiger80/keymaps/dmeiners88

## Keyboard Used
kbdfans/tiger80

## Link to product page (if applicable)
https://kbdfans.com/collections/tiger-lite

## Operating System
Windows 11 22H2 (Build 22621.1413)

## qmk doctor Output
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: /home/daniel/repos/qmk_firmware
Ψ Detected Linux (WSL, Ubuntu 20.04.6 LTS).
Ψ Git branch: tiger80-dmeiners88-keymap
Ψ Repo version: 0.20.1
Ψ - Latest tiger80-dmeiners88-keymap: 2023-03-10 14:26:25 +0100 (2f5b134899) -- Replace more magic numbers with constants
Ψ - Latest upstream/master: 2023-03-18 01:14:21 +0800 (e437560144) -- [Keyboard] Add yandrstudio/transition80 (#19790)
Ψ - Latest upstream/develop: 2023-03-18 01:21:53 +1100 (d8aec71e48) -- Remove trailing zeroes in info.json layouts (#20156)
Ψ - Common ancestor with upstream/master: 2023-03-18 01:14:21 +0800 (e437560144) -- [Keyboard] Add yandrstudio/transition80 (#19790)
Ψ - Common ancestor with upstream/develop: 2023-03-17 07:35:49 +1100 (1b3f7fcf7d) -- Add `qmk find` command, reuse logic for `qmk mass-compile`. (#20139)
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 9.2.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 6.3-20171130
Ψ Found dfu-programmer version 0.6.1
Ψ Found dfu-util version 0.9
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-01-03 19:29:26 +0000 --  (0062927e3)
Ψ - lib/chibios-contrib: 2023-01-11 16:42:27 +0100 --  (a224be15)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f)
Ψ QMK is ready to go
```

## Is AutoHotKey / Karabiner installed
- [x]  AutoHotKey (Windows)
- [ ] Karabiner (macOS)

## Other keyboard-related software installed
None besides qmk_toolbox.

## Additional Context
None

waffle87, 2023-03-17:
I presume it's related to #19975 and the RGB Light animations now being guarded behind `#ifdef RGB_DI_PIN`, which was moved to `info.json`.

dmeiners88, 2023-03-18:
Confirmed. Working for me again! Thanks for the quick fix!

closed 2023-03-19 by zvecr