qmk/qmk_firmware #20185: can define a key hold for a while and take effect?

lededev, 2023-03-19:
### Issue Description

Can I define a phyical key, press and hold for a while such as 4 seconds, and to take effect TG(1) do prevent accidental pressing.

lededev, 2023-03-19:
I noticed Mod-Tap, but my Shift Alt Ctrl keys are enough, not need more.
What I want is make some normal keys as Fn1 Fn2 ... layer keys, and do not lose their original functions, only more layers can extend total keys, Shift or Alt can not.

waffle87, 2023-03-19:
This sounds like a complicated [tap dance](https://docs.qmk.fm/#/feature_tap_dance).

lededev, 2023-03-19:
> This sounds like a complicated [tap dance](https://docs.qmk.fm/#/feature_tap_dance).

No, for example, let Capslock as Fn1, but Capslock still work as Cpaslock, so long as you press short time < 200ms.
Let Numberlock as Fn2, Numberlock still work as numberlock, but if hold Numberlock 4000ms, it will action as TG(2), I can define number pad keys as mouse keys, and no need hold numberlock to keep layer 2.
Let 'z' as Fn3 hold 200ms as Fn3, and left hand fingers 2 3 4 can reach wertsdfgxcvb and define them as layer 3 keys. Such as a left hand number pad, right hand can still hold mouse, very usefull when working with CAD or excel, which need both mouse and number pad operating.

lededev, 2023-03-19:
If I define Fn1..Fn10, I can easily make a 63-key keyboard have more than 600 keys.

lededev, 2023-03-21:
I find LT(layer, KC) can do [this](https://github.com/qmk/qmk_firmware/blob/master/docs/feature_layers.md#switching-and-toggling-layers-idswitching-and-toggling-layers), but in VIA, layer predefine LT() only space Fn1..3, I think I need hardcode my keyboard by calling LT(layer, KC) to define other keys.

closed 2023-03-21 by lededev