qmk/qmk_firmware #20206: [Bug] TT(layer) cannot toggle

willthong, 2023-03-21:
### Describe the Bug

Setting TT(1) and TT(2). Momentary working fine. Toggle not functioning at all.

Relevant lines from ``config.h``:

```
#define TAPPING_TOGGLE 2
#define TAPPING_TERM 200
```

I am on Arch Linux with ``avr-gcc 12.2.0-1`` - I believe that might be relevant because of [this closed issue](https://github.com/qmk/qmk_firmware/issues/17623). When I change ``OPT >= s`` to ``OPT >= 2`` in ``qmk_firmware/builddefs/common_rules.mk``, this issue is fixed but split keyboard functionality stops working.

Tagging @jypma and @kevinstadler who appear to have unresolved problems from the other, closed issue.

Any suggestions would be hugely appreciated!

### Keyboard Used

redox

### Operating System

Arch Linux

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

drashna, 2023-03-23:
Do you have LTO enabled, by chance?

willthong, 2023-04-03:
I've tried with both LTO enabled and disabled. When enabled, TT acts like a hold toggle instead, even though I have ```#define TAPPING_TOGGLE 2``` in my ```config.h```.

getreuer, 2023-04-23:
[The related thread](https://github.com/qmk/qmk_firmware/issues/17623) notes that changing avr-gcc version and/or optimization level appear to be workarounds, with suggestions that the root cause is "an optimization related bug." OTOH, a common reason of course for optimization bugs is that the code has undefined behavior, not necessarily that the compiler is wrong. So we should check.

There's a unit test [tests/basic/test_action_layer.cpp](https://github.com/qmk/qmk_firmware/blob/master/tests/basic/test_action_layer.cpp) that exercises `TT` keys. I'd like to try running it with Valgrind or ASan. I know I can run the test with `make test:basic`, but I don't follow the Makefiles enough to see how that works. 

Does anyone know how to get a test executable or build a test with different options?