qmk/qmk_firmware #20228: [Feature Request] Adding support for the STM32H750 family MCUs?

JunASAKA, 2023-03-22:
### Feature Request Type

- [X] Core functionality
- [X] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

Hey guys,
I am trying to port the QMK to STM32H750 which already has a `stm32_registry.h` in the ChibiOS repo. I am wondering how can I do to compile the firmware for it?
When I tried to compile it after adding some configuration code, many errors occurred. So how could I build the firmware for other MCUs like STM32H750 which already have `stm32_registry.h` existed.

Thanks and Regards,
Jun ASAKA.

drashna, 2023-03-23:
This isn't comprehensive, but the files that it would need to be added to: 

* data/schemas/keyboard.jsonschema
* lib/python/qmk/constants.py
* platforms/chibios/mcu_selection.mk

ANd I think a generic board needs to be added to platforms/chibios/boards as well.

tzarc, 2023-03-23:
You should be able to manually specify all the parameters in the keyboard's `rules.mk`, much like `kinesis/kint41` does.

JunASAKA, 2023-03-23:
I found that there are some problem in `stm32_clock_init();`
the function stuck at here:
`while ((RCC->CR & RCC_CR_HSERDY) == 0);`
...
`while ((RCC->CR & rdymask) != rdymask);`
...
`while ((RCC->CFGR & RCC_CFGR_SWS_Msk) != (STM32_SW << RCC_CFGR_SWS_Pos));`
And when remove these three `while` loops, the board would booted up normally. So I think there are some issues when initialising the clocks.

JunASAKA, 2023-03-24:
> I found that there are some problem in `stm32_clock_init();`(lib/chibios/os/hal/ports/STM32/STM32H7xx/hal_lld.c) the function stuck at here: `while ((RCC->CR & RCC_CR_HSERDY) == 0);` ... `while ((RCC->CR & rdymask) != rdymask);` ... `while ((RCC->CFGR & RCC_CFGR_SWS_Msk) != (STM32_SW << RCC_CFGR_SWS_Pos));` And when remove these three `while` loops, the board would booted up normally. So I think there are some issues when initialising the clocks.

Oh yee. I just fixed it by removing `#define STM32_HSE_BYPASS` from board.h which made the HSE failed. And I think this should be customised according to the boards.  
So what should I do to make a PR to support STM32H750?

Regards,
Jun ASAKA.

tzarc, 2023-03-29:
At this point your best bet is to discuss directly with the ChibiOS maintainers -- they're pretty responsive on their forums and/or Discord.

tzarc, 2023-06-28:
Feel free to tackle the same thing as #21352 once ChibiOS itself is fixed up.

closed 2023-06-28 by tzarc