qmk/qmk_firmware #20233: [Feature Request] Custom Bluetooth driver

Goshin, 2023-03-22:
### Feature Request Type

- [X] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [X] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

Currently, `BLUETOOTH_DRIVER` can be set to either `BluefruitLE`, `RN42`, or`custom`. For `custom` I suppose it, similar to other drivers, would allow developers to implement their own communication with other Bluetooth accessory boards in their keyboards/keymaps. 

https://github.com/qmk/qmk_firmware/blob/bacec14073b2e897d5a52caf12de5a6a1f7b4078/builddefs/common_features.mk#L886-L909

However, the current implementation only considers the embedded drivers.

https://github.com/qmk/qmk_firmware/blob/5ace174cd3894c485157773c73a1cb40999cdd7b/drivers/bluetooth/bluetooth.c#L40-L47

So I suggest we define the Bluetooth-related functions in `drivers/bluetooth/bluetooth.c` as weak symbols, which allows developers to override these functions for their keyboards/keymaps.

```diff
@@ -23,7 +23,7 @@
 #    include "rn42.h"
 #endif

-void bluetooth_init(void) {
+__attribute__((weak)) void bluetooth_init(void) {
 #if defined(BLUETOOTH_BLUEFRUIT_LE)
     bluefruit_le_init();
 #elif defined(BLUETOOTH_RN42)
@@ -31,13 +31,13 @@ void bluetooth_init(void) {
 #endif
 }

-void bluetooth_task(void) {
+__attribute__((weak)) void bluetooth_task(void) {
 #if defined(BLUETOOTH_BLUEFRUIT_LE)
     bluefruit_le_task();
 #endif
 }

-void bluetooth_send_keyboard(report_keyboard_t *report) {
+__attribute__((weak)) void bluetooth_send_keyboard(report_keyboard_t *report) {
 #if defined(BLUETOOTH_BLUEFRUIT_LE)
     bluefruit_le_send_keyboard(report);

...
```

---

Alternatively, we can exclude `drivers/bluetooth/bluetooth.c` in compiling when`BLUETOOTH_DRIVER` is 'custom'. In this way, developers would have to provide their own implementations for all the Bluetooth functions.

```diff
diff --git a/builddefs/common_features.mk b/builddefs/common_features.mk
index d9130b5338..9ef926f5a4 100644
@@ -884,7 +884,10 @@ ifeq ($(strip $(BLUETOOTH_ENABLE)), yes)
     OPT_DEFS += -DBLUETOOTH_ENABLE
     NO_USB_STARTUP_CHECK := yes
     COMMON_VPATH += $(DRIVER_PATH)/bluetooth
-    SRC += outputselect.c bluetooth.c
+    SRC += outputselect.c
+    ifneq ($(strip $(BLUETOOTH_DRIVER)), custom)
+        SRC += bluetooth.c
+    endif

     ifeq ($(strip $(BLUETOOTH_DRIVER)), BluefruitLE)
         OPT_DEFS += -DBLUETOOTH_BLUEFRUIT_LE -DHAL_USE_SPI=TRUE 
```

---

Other changes that we might need to make:
- Documentation
- For a more consistent style, make the default implementations blank and let the drivers override them.

drashna, 2023-03-23:
@fauxpark thoughts?

fauxpark, 2023-03-23:
The Bluetooth API has not been fully solidified, this will be sorted out some time in the future.

fauxpark, 2023-06-24:
This has been somewhat resolved as of https://github.com/qmk/qmk_firmware/pull/20897

closed 2023-06-24 by fauxpark