qmk/qmk_firmware #20239: Underglow doesn't work with WS2812B, but simple code using FastLED works

linhx, 2023-03-23:
### Issue Description

I'm creating a custom Macropad, using QMK. Everything is working fine except the underglow RGB.
QMK version: 1.1.1 (commit 462ab3c4d23615b0a80a19b18ab7c69cf8904313)
My components:

- 4 LEDs WS2812B
- ProMicro clone

Here are how I wire and config.
[image]
[image]
[image]

I also configed the [bit timings](https://docs.qmk.fm/#/ws2812_driver?id=adjusting-bit-timings) follow the [datasheet](https://cdn-shop.adafruit.com/datasheets/WS2812B.pdf) but still no luck.

```c
#define RGB_DI_PIN C6
#define RGBLED_NUM 4
#define WS2812_TRST_US 80
#define WS2812_TIMING 1250
#define WS2812_T0H 400
#define WS2812_T1H 800
#define RGBLIGHT_EFFECT_BREATHING
#define RGBLIGHT_DEFAULT_MODE RGBLIGHT_EFFECT_BREATHING + 1
```

---

But the LEDs work with FastLED:

```c
#include "FastLED.h"

#define NUM_LEDS 4
#define DATA_PIN 5

CRGB leds[NUM_LEDS];

void setup() { 
      FastLED.addLeds<WS2812B, DATA_PIN, GRB>(leds, NUM_LEDS);
}

void loop() {
  leds[0] = CRGB::Green;
  leds[1] = CRGB::Green;
  leds[2] = CRGB::Green;
  leds[3] = CRGB::Green;
  FastLED.show();
  delay(500);
  // Now turn the LED off, then pause
  leds[0] = CRGB::Black;
  leds[1] = CRGB::Black;
  leds[2] = CRGB::Black;
  leds[3] = CRGB::Black;
  FastLED.show();
  delay(500);
}
```

drashna, 2023-03-23:
set the led number to 8, and try?

IIRC< there is a timing issue with less than 8 leds.

drashna, 2023-03-30:
It was mentioned on discord recently that there may be a bug with the rgblight code.  I think it was this commit: https://github.com/qmk/qmk_firmware/commit/f6baf916a9ac14cd150f65054573cd1a0e732c5a

Reverting it for now may help.

linhx, 2023-04-01:
@drashna unfortunaly it still doesn't work

sigprof, 2023-04-01:
> ```c
> #define RGBLIGHT_DEFAULT_MODE RGBLIGHT_EFFECT_BREATHING + 1
> ```

This code is wrong — probably you meant
```c
#define RGBLIGHT_DEFAULT_MODE RGBLIGHT_MODE_BREATHING + 1
```
The code on the screenshot has a similar error; there is also the `RGBLIGHT_DEFAULT_VAR` typo there — that define does not do anything.

closed 2023-08-01 by github-actions[bot]