qmk/qmk_firmware #20258: [RFC, PoC, etc.] GPIO abstractions

tzarc, 2023-03-25:
### Issue Description

`<placeholder>`

elpekenin, 2023-03-26:
I'd like to help with this. Couple things:

1. Where should this code be located? Im thinking on a `pin` folder where `pin.h` defines the types and macros and `pin.c` "merges" all the types on the `vtables` array. And several, rather small, `.c` files with type-specific logic
    - When using basic MCU pins, the way to interact with a pin would be different (ChibiOS vs VUSB for example), so those definitions should be defined on `platforms`.... As far as i know
    - Other "pins" should be able to use QMK abstractions to talk with IC's over SPI or whatever, so that should probably live on `quantum`

1. We would like some extra checks, wouldn't we?
    - Eg, `ifdef MCU_RP2040` only the last GPIOs can be used for analog operation, so we could have a `if (pin->mcu_pin > 25U)` (wrote a random number here) to do `dprintf("Analog reading not supported on pin %d\n", pin->mcu_pin)`
    - Is this handled already, to some extend?

sigprof, 2023-03-26:
```c
typedef struct pin_impl_t {
	pintype_t type;
	union {
		uint16_t mcu_pin;
```
This would probably end up being something like `platform_mcu_pin_t mcu_pin`, with `platform_mcu_pin_t` being defined as `ioline_t` for ChibiOS, `uint8_t` for AVR, or whatever it is called in RIOT.

Note that `ioline_t` in ChibiOS is defined by the HAL port code, so the underlying type may still vary.  Currently all ChibiOS ports we care about use `uint32_t` (HT32 uses `uintptr_t`, but it's `uint32_t` under the hood anyway); AVR ports have a struct there, but QMK does not support those ports.  In the STM32 implementation the `ioline_t` value actually embeds the pointer to the GPIO register block inside (the code assumes that 4 low bits of that pointer are 0, and uses those bits to store the pad number); direct usage of the GPIO number on RP2040 is a really special case.

With `uint32_t` there the `pintype_t type` field would occupy the same space as `gpio_pin_vtable_t *vtable` due to alignment; so the space optimization would really affect only the AVR case (but that would also be the case where it's the most needed — on proper MCUs that `const` data won't be copied to RAM anyway).  Having the vtable pointer there directly would also allow the linker to discard a GPIO implementation that ends up completely unused, but that optimization may not be really important (the proper way is to leave the unused implementations disabled).  Unfortunately, there seems to be no way to discard methods that are never actually used anywhere (apart from adding config options for them too, but that would be lots of ifdefs scattered around all GPIO implementations).

Also with some casts we could use separate structs for pin types instead of a single union for all types, and save some bytes (because the structs won't be padded to the maximum possible size).

sigprof, 2023-03-26:
> * Eg, `ifdef MCU_RP2040` only the last GPIOs can be used for analog operation

Currently the ChibiOS analog driver code would just silently return 0 if you call `analogReadPin()` for an unsupported pin, because `pinToMux()` would return an invalid value pointing to a nonexistent ADC.  The AVR code apparently attempts to select some nonexistent ADC input channel which gives 0V too (but the conversion would still be performed and may read some noise).

Making this code work with the new GPIO abstraction would require either rewriting parts of it (things like `case F0:` won't work), or making the code dig into the abstraction (check for `pin->type == MCU_PIN`, then extract the `mcu_pin` value and work from there).  Another possible solution is adding the `analog_read` method to GPIO methods — this way you could potentially implement support for external ADCs, but in this case a config option around that method would definitely be required, otherwise the analog support would always be referenced.

elpekenin, 2023-03-26:
> check for pin->type == MCU_PIN, then extract the mcu_pin value and work from there

Yeah, was thinking of `vtable`s containing that, so that other abstractions can also be used for analog reads + at that point we can (hopefully?) perform a guard clause like the one i wrote, on the MCU-pin function, before calling the underlying implementation

tzarc, 2023-03-27:
So the casting would result in something like:

```c
typedef struct pin_base_t {
	pin_type_t type;
} pin_base_t;
```
```c
typedef struct pin_chibios_t {
	pin_base_t base;
	ioline_t line;
} pin_chibios_t;
```
```c
typedef struct pin_avr_t {
	pin_base_t base;
	uint8_t pin;
} pin_avr_t;
```
```c
// #define GP29 29U
static const pin_chibios_t pin_gp29 = {.base = {.type = MCU_PIN}, .line = 29U};
#define GP29 ((const struct pin_base_t*)&pin_gp29)
```

elpekenin, 2023-04-08:
BTW im making some experiments over [here](https://github.com/elpekenin/qmk_firmware/tree/pin-playground)(still minimal, it compiles but havent flashed anything yet).

Sharing in case someone wants to provide some feedback, help with it, and trying to avoid other people duplicating work

elpekenin, 2023-04-24:
Made some progress, would love some feedback 🤓 

https://github.com/elpekenin/qmk_firmware/commit/0233218999d11936524345003d0cfeacacaf0994

closed 2023-08-24 by github-actions[bot]