qmk/qmk_firmware #20284: [Bug]

quinn, 2023-03-28:
### Describe the Bug

When I try to run the qmk flash command, I get an error "NoBackendFound":

```qmk flash $HOME/Downloads/v4_us_v1.4.bin                             
Flashing binary firmware...
Please reset your keyboard into bootloader mode now!
Press Ctrl-C to exit.

<class 'usb.core.NoBackendError'>
☒ No backend available
Traceback (most recent call last):
  File "/home/quinn/.local/lib/python3.10/site-packages/milc/milc.py", line 528, in __call__
    return self.__call__()
  File "/home/quinn/.local/lib/python3.10/site-packages/milc/milc.py", line 533, in __call__
    return self._subcommand(self)
  File "/home/quinn/qmk_firmware/lib/python/qmk/decorators.py", line 27, in wrapper
    return func(*args, **kwargs)
  File "/home/quinn/qmk_firmware/lib/python/qmk/decorators.py", line 47, in wrapper
    return func(*args, **kwargs)
  File "/home/quinn/qmk_firmware/lib/python/qmk/cli/flash.py", line 93, in flash
    return _flash_binary(cli.args.filename, cli.args.mcu)
  File "/home/quinn/qmk_firmware/lib/python/qmk/cli/flash.py", line 59, in _flash_binary
    err, msg = flasher(mcu, filename)
  File "/home/quinn/qmk_firmware/lib/python/qmk/flashers.py", line 200, in flasher
    bl, details = _find_bootloader()
  File "/home/quinn/qmk_firmware/lib/python/qmk/flashers.py", line 88, in _find_bootloader
    dev = _find_usb_device(vid_hex, pid_hex)
  File "/home/quinn/qmk_firmware/lib/python/qmk/flashers.py", line 71, in _find_usb_device
    return usb.core.find(idVendor=vid_hex, idProduct=pid_hex)
  File "/home/quinn/.local/lib/python3.10/site-packages/usb/core.py", line 1309, in find
    raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend available
```

### Keyboard Used

passing binary file directly

### Link to product page (if applicable)

https://www.keychron.com/products/keychron-v4-qmk-custom-mechanical-keyboard

### Operating System

Ubuntu 22.04

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: /home/quinn/qmk_firmware
Ψ Detected Linux (Ubuntu 22.04.2 LTS).
Ψ Git branch: master
Ψ Repo version: 0.20.1
Ψ - Latest master: 2023-03-29 01:24:53 +0900 (a6b2b1ca41) -- Add new keyboard Tightwriter (#20136)
Ψ - Latest upstream/master: 2023-03-29 01:24:53 +0900 (a6b2b1ca41) -- Add new keyboard Tightwriter (#20136)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2023-03-29 01:24:53 +0900 (a6b2b1ca41) -- Add new keyboard Tightwriter (#20136)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.3.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 6.3-20171130
Ψ Found dfu-programmer version 0.6.1
Ψ Found dfu-util version 0.9
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-01-03 19:29:26 +0000 --  (0062927e3)
Ψ - lib/chibios-contrib: 2023-01-11 16:42:27 +0100 --  (a224be15)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

tzarc, 2023-03-29:
Can you try running `./util/qmk_install.sh`? Looks like a dependency is missing.

619frank, 2024-12-27:
I faced the same error. It seems that on Windows, we need to follow the steps mentioned in this Stack Overflow post:

[PyUSB on Windows: No backend available](https://stackoverflow.com/questions/13773132/pyusb-on-windows-no-backend-available).

Following these steps resolved the issue for me.