qmk/qmk_firmware #20378: [Bug] Preonic compilation warning "Invalid LAYOUT macro" because of key having multiple matrix positions

brauner, 2023-04-09:
### Describe the Bug

When compiling firmware for the Preonic and my upstream keymap via:

```
qmk compile -kb preonic/rev3_drop -km brauner
```

I consistently get the following warnings:

```
QMK Firmware 0.19.5
☒ Invalid LAYOUT macro in keyboards/preonic/rev3_drop/rev3_drop.h: Key k45 in macro LAYOUT_preonic_1x2uC has multiple matrix positions ([9, 0], [9, 5])
☒ Invalid LAYOUT macro in keyboards/preonic/rev3_drop/rev3_drop.h: Key k47 in macro LAYOUT_preonic_1x2uR has multiple matrix positions ([9, 0], [9, 1])
☒ Invalid LAYOUT macro in keyboards/preonic/rev3_drop/rev3_drop.h: Key k44 in macro LAYOUT_preonic_1x2uL has multiple matrix positions ([9, 4], [9, 5])
☒ Invalid LAYOUT macro in keyboards/preonic/rev3_drop/rev3_drop.h: Key k44 in macro LAYOUT_preonic_2x2u has multiple matrix positions ([9, 4], [9, 5])
☒ Invalid LAYOUT macro in keyboards/preonic/rev3_drop/rev3_drop.h: Key k46 in macro LAYOUT_preonic_2x2u has multiple matrix positions ([9, 0], [9, 1])
Making preonic/rev3_drop with keymap brauner

☒ Invalid LAYOUT macro in keyboards/preonic/rev3_drop/rev3_drop.h: Key k45 in macro LAYOUT_preonic_1x2uC has multiple matrix positions ([9, 0], [9, 5])
☒ Invalid LAYOUT macro in keyboards/preonic/rev3_drop/rev3_drop.h: Key k47 in macro LAYOUT_preonic_1x2uR has multiple matrix positions ([9, 0], [9, 1])
☒ Invalid LAYOUT macro in keyboards/preonic/rev3_drop/rev3_drop.h: Key k44 in macro LAYOUT_preonic_1x2uL has multiple matrix positions ([9, 4], [9, 5])
☒ Invalid LAYOUT macro in keyboards/preonic/rev3_drop/rev3_drop.h: Key k44 in macro LAYOUT_preonic_2x2u has multiple matrix positions ([9, 4], [9, 5])
☒ Invalid LAYOUT macro in keyboards/preonic/rev3_drop/rev3_drop.h: Key k46 in macro LAYOUT_preonic_2x2u has multiple matrix positions ([9, 0], [9, 1])
```

I've waited for a couple of weeks now to see whether this warning would disappear in master but so far it hasn't.

### Keyboard Used

preonic/rev3_drop

### Operating System

Linux

### qmk doctor Output

```text
> qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: /home/brauner/src/git/qmk_firmware
Ψ Detected Linux (Ubuntu 22.04.2 LTS).
⚠ Missing or outdated udev rules for 'atmel-dfu' boards. Run 'sudo cp /home/brauner/src/git/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'kiibohd' boards. Run 'sudo cp /home/brauner/src/git/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'stm32-dfu' boards. Run 'sudo cp /home/brauner/src/git/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'apm32-dfu' boards. Run 'sudo cp /home/brauner/src/git/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'gd32v-dfu' boards. Run 'sudo cp /home/brauner/src/git/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'bootloadhid' boards. Run 'sudo cp /home/brauner/src/git/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbasploader' boards. Run 'sudo cp /home/brauner/src/git/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbtinyisp' boards. Run 'sudo cp /home/brauner/src/git/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'md-boot' boards. Run 'sudo cp /home/brauner/src/git/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Detected ModemManager without the necessary udev rules. Please either disable it or set the appropriate udev rules if you are using a Pro Micro.
⚠ Missing or outdated udev rules for 'caterina' boards. Run 'sudo cp /home/brauner/src/git/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'hid-bootloader' boards. Run 'sudo cp /home/brauner/src/git/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
Ψ Git branch: master
Ψ Repo version: 0.19.5
Ψ - Latest master: 2023-04-07 17:59:13 -0700 (2c375e6478) -- [Keyboard] Add audio support to Adafruit MacroPad RP2040 (#20353)
Ψ - Latest upstream/master: 2023-04-07 17:59:13 -0700 (2c375e6478) -- [Keyboard] Add audio support to Adafruit MacroPad RP2040 (#20353)
Ψ - Latest upstream/develop: 2022-12-27 20:29:52 +0000 (40fddcb1e5) -- Merge remote-tracking branch 'origin/master' into develop
Ψ - Common ancestor with upstream/master: 2023-04-07 17:59:13 -0700 (2c375e6478) -- [Keyboard] Add audio support to Adafruit MacroPad RP2040 (#20353)
Ψ - Common ancestor with upstream/develop: 2022-12-27 20:29:52 +0000 (40fddcb1e5) -- Merge remote-tracking branch 'origin/master' into develop
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.3.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 6.3-20171130
Ψ Found dfu-programmer version 0.6.1
Ψ Found dfu-util version 0.9
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-01-03 19:29:26 +0000 --  (0062927e3)
Ψ - lib/chibios-contrib: 2023-01-11 16:42:27 +0100 --  (a224be15)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

fauxpark, 2023-04-09:
It's been fixed in develop.

brauner, 2023-04-09:
Thanks! Switched to `develop` and it works fine now!

closed 2023-04-09 by brauner