qmk/qmk_firmware #20415: [Bug] process_record_user SEND_STRING doesn't work for multiple cases

jazark, 2023-04-12:
### Describe the Bug

Setting `phrase1` & `phrase2` from `process_record_user` does not seem to be firing for `phrase2`.
Phrase1 works fine, I am able to map it to any key, however, the case logic for `phrase2` isn't working correctly.

I am following the example from master: https://github.com/qmk/qmk_firmware/blob/master/docs/custom_quantum_functions.md#example-process_record_user-implementation

Here is what my KB layout looks like:

> #include QMK_KEYBOARD_H
> #define _LAYER0 0
> #define _LAYER1 1
> // #define _GAMING 2
> 
> enum custom_keycodes {
>     phrase1,
> 	phrase2,
> };
> 
> const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
> 			/* Base layer
> 		* ,--------------------------------------------------------------------------------------------------.
> 		* | Esc |  1  |  2  |  3  |  4  |  5  |  6  |  7  |  8  |  9  |  0  |  -  |  =  |  Backspace  | `    |
> 		* |-------------------------------------------------------------------------------------------+------+
> 		* | Tab    |  Q  |  W  |  E  |  R  |  T  |  Y  |  U  |  I  |  O  |  P  |  [  |  ]  |    \     | Del  |
> 		* |-------------------------------------------------------------------------------------------+------+
> 		* | MO(_FN2) |  A  |  S  |  D  |  F  |  G  |  H  |  J  |  K  |  L  |  ;  |  '  |     Enter    | PgUp |
> 		* |-------------------------------------------------------------------------------------------+------+
> 		* | Shift      |  Z  |  X  |  C  |  V  |  B  |  N  |  M  |  ,  |  .  |  /  |   Shift    | Up  | PgDn |
> 		* +-------------------------------------------------------------------------┬---┬-------------+------+
> 		* | LCtrl | LAlt | LGUI |               Space                 | RGUI | RAlt |   | Left  | Dn  | Rght |
> 		* `-------------------------------------------------------------------------┘   └-------------+------´
> 		*/
> 		[_LAYER0] = LAYOUT_65_ansi_blocker( /* Base */
> 			KC_ESC,       KC_1,     KC_2,     KC_3,  KC_4,  KC_5,  KC_6,    KC_7,  KC_8,    KC_9,     KC_0,     KC_MINS,  KC_EQL,  KC_BSPC,   KC_DEL,\
> 		    KC_TAB,        KC_Q,     KC_W,     KC_E,  KC_R,  KC_T,  KC_Y,    KC_U,  KC_I,    KC_O,     KC_P,     KC_LBRC,  KC_RBRC, KC_BACKSLASH, KC_PGUP,\
> 			LT(1,_______),	   KC_A,     KC_S,     KC_D,  KC_F,  KC_G,  KC_H,    KC_J,  KC_K,    KC_L,     KC_SCLN,  KC_QUOT,           KC_ENT,    KC_PGDN,\
> 		    KC_LSFT,       KC_Z,     KC_X,     KC_C,  KC_V,  KC_B,  KC_N,    KC_M,  KC_COMM, KC_DOT,   KC_SLSH,  KC_RSFT,           KC_UP,     KC_END,\
> 		    KC_LCTL,       KC_LGUI,  KC_LALT,                KC_SPC,                KC_RALT,           MO(1),    KC_LEFT,           KC_DOWN,   KC_RIGHT),
> 		[_LAYER1] = LAYOUT_65_ansi_blocker( /* FN */
> 			KC_ESC,       KC_F1,    KC_F2,   KC_F3,  KC_F4,   KC_F5,   KC_F6,   KC_F7,   KC_F8,   KC_F9,   KC_F10,   KC_F11,  KC_F12,  KC_DEL,   KC_HOME,\
> 		    KC_GRV,       RGB_TOG,  RGB_MOD, RGB_HUI,RGB_HUD, RGB_SAI, RGB_SAD, RGB_VAI, RGB_VAD, KC_TRNS, KC_PSCR, KC_SLCT,  KC_PAUS, QK_REBOOT,    KC_PGUP,\
> 			_______,       RGB_SPI,  RGB_SPD, KC_TRNS,KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS,  KC_TRNS,          EE_CLR,  KC_PGDN,\
> 		    KC_LSFT,       KC_TRNS,  KC_TRNS, KC_TRNS,KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, phrase2, KC_TRNS,  AS_TOGG,          KC_VOLU,  KC_MUTE,\
> 		    KC_TRNS,       KC_TRNS,  KC_TRNS,                  phrase1,                   KC_GRV,          KC_TRNS,  KC_MPRV,          KC_VOLD,  KC_MNXT),

Below is the `process_record_user` function:

```
bool process_record_user(uint16_t keycode, keyrecord_t *record) {
  switch (keycode) {
    case phrase1:
      if (record->event.pressed) {
		SEND_STRING("Hello!");
      }
	  return true;
    case phrase2:
      if (record->event.pressed) {
		SEND_STRING("World");
      }
      return true; // Let QMK send the enter press/release events
    default:
      return true; // Process all other keycodes normally
  }
}
```

### Keyboard Used

kbdfans/kbd67/mkiirgb/v3

### Link to product page (if applicable)

https://www.monstargears.com/51/?idx=503

### Operating System

Wndows 10

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

tzarc, 2023-04-12:
you need to change:
```
enum custom_keycodes {
    phrase1,
	phrase2,
};
```
to:
```
enum custom_keycodes {
    phrase1 = QK_USER_0,
	phrase2,
};
```

jazark, 2023-04-12:
@tzarc Thank you dude, this worked and fixed the problem on my end!