qmk/qmk_firmware #20448: [Bug] SEND_STRING does not support X_NUM or X_NUM_LOCK

TurtleHunter, 2023-04-14:
### Describe the Bug

I have a specific use case where I need to send a string containing two taps of Num Lock.

```c
SEQ_ONE_KEY(KC_R) {
	SEND_STRING_DELAY(SS_TAP(X_LCTL) SS_TAP(X_LCTL) SS_DOWN(X_NUM) SS_UP(X_NUM) SS_DOWN(X_NUM) SS_UP(X_NUM), 10);
}
```
also (preferred syntax): 
```c
SEND_STRING(SS_TAP(X_LCTL) SS_TAP(X_LCTL) SS_TAP(X_NUM) SS_TAP(X_NUM);
```
as well as using `X_NUM_LOCK` gives:
```
Compiling: quantum/keymap_introspection.c                                                          In file included from quantum/keymap_introspection.c:5:
./keyboards/massdrop/alt/keymaps/TurtleHunter/keymap.c: In function 'matrix_scan_user':
./keyboards/massdrop/alt/keymaps/TurtleHunter/keymap.c:78:1: error: \x used with no following hex digits
   78 |    SEND_STRING_DELAY(SS_TAP(X_LCTL) SS_TAP(X_LCTL) SS_DOWN(X_NUM) SS_UP(X_NUM) SS_DOWN(X_NUM) SS_UP(X_NUM), 10);
      | ^  ~~~
./keyboards/massdrop/alt/keymaps/TurtleHunter/keymap.c:78:1: error: \x used with no following hex digits
./keyboards/massdrop/alt/keymaps/TurtleHunter/keymap.c:78:1: error: \x used with no following hex digits
./keyboards/massdrop/alt/keymaps/TurtleHunter/keymap.c:78:1: error: \x used with no following hex digits
 [ERRORS]
 |
 |
 |
make[1]: *** [builddefs/common_rules.mk:359: .build/obj_massdrop_alt_TurtleHunter/quantum/keymap_introspection.o] Error 1
Make finished with errors
make: *** [Makefile:414: massdrop/alt:TurtleHunter] Error 1
```

---

The following was able to compile, but does not work for my use case:
```c
SEQ_ONE_KEY(KC_R) {
    SEND_STRING_DELAY(SS_TAP(X_LCTL) SS_TAP(X_LCTL) SS_DOWN(X_NUM) SS_UP(X_NUM) SS_DOWN(X_NUM) SS_UP(X_NUM), 10);
}
```

### Keyboard Used

massdrop/alt

### Operating System

Windows 11

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: C:/Users/Samuel/qmk_firmware
Ψ Detected Windows 10 (10.0.22624).
Ψ Git branch: master
Ψ Repo version: 0.18.16
⚠ Git has unstashed/uncommitted changes.
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.1.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2022-06-11 09:09:45 +0000 --  (f836d24b0)
Ψ - lib/chibios-contrib: 2022-07-27 10:46:15 +0200 --  (d03aa9cc)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/pico-sdk: 2022-05-17 19:19:01 +0200 --  (07edde8)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

TurtleHunter, 2023-04-14:
I just thought to review the definitions for keycodes, as Send_String uses X_ keycodes instead of KC_
in `qmk_firmware\qmk_firmware\quantum\send_string\send_string_keycodes.h` line 43:
```c
#define X_NLCK X_NUMLOCK
```
this does not match the documentation for Send_String, which says to simply change out KC for X. The KC keycode for Num Lock is `KC_NUM_LOCK` or `KC_NUM`
[Send_String documentation](https://docs.qmk.fm/#/feature_send_string)
[Basic Keycodes documentation](https://docs.qmk.fm/#/keycodes_basic?id=lock-keys)
[Keycodes definition](https://github.com/qmk/qmk_firmware/blob/master/docs/keycodes.md)

fauxpark, 2023-04-15:
Your repo is out of date.
[image]

closed 2023-04-15 by fauxpark