qmk/qmk_firmware #20449: unregister_code(KC_LOPTN);

Lorn888, 2023-04-14:
### Issue Description

When trying to compile for my keychron k3pro I get the following error (error below) but if I use any other key than KC_LOPTN it works fine 

Ψ Compiling keymap with gmake --jobs=1 keychron/k3_pro/ansi/rgb:lorn888

QMK Firmware 0.14.29
Making keychron/k3_pro/ansi/rgb with keymap lorn888

arm-none-eabi-gcc (15:8-2019-q3-1+b1) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text    data     bss     dec     hex filename
      0   60070       0   60070    eaa6 keychron_k3_pro_ansi_rgb_lorn888.bin

Compiling: quantum/keymap_introspection.c                                                          In file included from quantum/keymap_introspection.c:5:
./keyboards/keychron/k3_pro/ansi/rgb/keymaps/lorn888/keymap.c: In function 'process_record_user':
./keyboards/keychron/k3_pro/ansi/rgb/keymaps/lorn888/keymap.c:73:23: error: unsigned conversion from 'int' to 'uint8_t' {aka 'unsigned char'} changes value from '32256' to '0' [-Werror=overflow]
         register_code(KC_LOPTN);
                       ^~~~~~~~
./keyboards/keychron/k3_pro/ansi/rgb/keymaps/lorn888/keymap.c:76:25: error: unsigned conversion from 'int' to 'uint8_t' {aka 'unsigned char'} changes value from '32256' to '0' [-Werror=overflow]
         unregister_code(KC_LOPTN);
                         ^~~~~~~~
./keyboards/keychron/k3_pro/ansi/rgb/keymaps/lorn888/keymap.c:84:21: error: unsigned conversion from 'int' to 'uint8_t' {aka 'unsigned char'} changes value from '32256' to '0' [-Werror=overflow]
       register_code(KC_LOPTN);
                     ^~~~~~~~
./keyboards/keychron/k3_pro/ansi/rgb/keymaps/lorn888/keymap.c:87:23: error: unsigned conversion from 'int' to 'uint8_t' {aka 'unsigned char'} changes value from '32256' to '0' [-Werror=overflow]
       unregister_code(KC_LOPTN);
                       ^~~~~~~~
./keyboards/keychron/k3_pro/ansi/rgb/keymaps/lorn888/keymap.c:98:23: error: unsigned conversion from 'int' to 'uint8_t' {aka 'unsigned char'} changes value from '32256' to '0' [-Werror=overflow]
         register_code(KC_LOPTN);
                       ^~~~~~~~
./keyboards/keychron/k3_pro/ansi/rgb/keymaps/lorn888/keymap.c:101:25: error: unsigned conversion from 'int' to 'uint8_t' {aka 'unsigned char'} changes value from '32256' to '0' [-Werror=overflow]
         unregister_code(KC_LOPTN);
                         ^~~~~~~~
./keyboards/keychron/k3_pro/ansi/rgb/keymaps/lorn888/keymap.c:109:21: error: unsigned conversion from 'int' to 'uint8_t' {aka 'unsigned char'} changes value from '32256' to '0' [-Werror=overflow]
       register_code(KC_LOPTN);
                     ^~~~~~~~
./keyboards/keychron/k3_pro/ansi/rgb/keymaps/lorn888/keymap.c:112:23: error: unsigned conversion from 'int' to 'uint8_t' {aka 'unsigned char'} changes value from '32256' to '0' [-Werror=overflow]
       unregister_code(KC_LOPTN);
                       ^~~~~~~~
cc1: all warnings being treated as errors
 [ERRORS]
 | 
 | 
 | 
gmake[1]: *** [builddefs/common_rules.mk:360: .build/obj_keychron_k3_pro_ansi_rgb_lorn888/quantum/keymap_introspection.o] Error 1
Make finished with errors
gmake: *** [Makefile:414: keychron/k3_pro/ansi/rgb:lorn888] Error 1 

`/* Copyright 2023 @ Keychron (https://www.keychron.com)
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 2 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 */

#include QMK_KEYBOARD_H
[... 83 lines cut ...]
        unregister_code(KC_LOPTN);
        unregister_code(KC_TRNS);
      }
      g_pressed = false;
    }
  } else if (keycode == KC_A && g_pressed) {
    if (record->event.pressed && (timer_elapsed(i_timer_start) > TAPPING_TERM_G)) {
      register_code(KC_TRNS);
      register_code(KC_LOPTN);
      register_code(KC_2);
      unregister_code(KC_2);
      unregister_code(KC_LOPTN);
      unregister_code(KC_TRNS);
    }
    g_pressed = false;
  }
  return true;
}

void matrix_scan_user(void) {
  if (g_pressed && (timer_elapsed(i_timer_start) >= TAPPING_TERM_G)) {
    g_pressed = false;
  }
}
`

elpekenin, 2023-04-14:
Using code blocks (ie: wrapping code in triple backticks) would have made the message waaay easier to read

The issue is that you are trying to use a custom keycode (ie: something that the computer doesnt understand, and that keychron has defined to trigger some custom logic on the keyboard), which `register_code` which is meant as "send this to the computer"(for keys, numbers and stuff like that)

You are also using `unregister_code(KC_TRNS)` which is useless for the same reason (transparent keys are a thing on QMK to work with layers, not something the computer will understand)

Would  you mind explaining what are you trying to achieve with this weird block of code? (also, perhaps you'd get faster answers on QMK's discord server)

sigprof, 2023-04-15:
(I edited your message to add triple backticks at the appropriate places, so that the code is readable.)

The `KC_LOPTN`, `KC_ROPTN`, `KC_LCMMD`, `KC_RCMMD` keycodes were introduced by Keychron in order to make them appear as “Option” and “Command” in VIA; they are mostly equivalent to the already existing basic keycodes `KC_LOPT`, `KC_ROPT`, `KC_LCMD`, `KC_RCMD` (which are actually aliases for `KC_LALT`, `KC_RALT`, `KC_LGUI`, `KC_RGUI`), except those Keychron-specific codes are not basic keycodes and cannot be used anywhere except in the keymap.  So the simplest solution is to forget that those Keychron-specific keycodes exist and replace them with basic keycodes like `KC_LOPT`.

And apparently you are trying to implement something like dual-role keys with distinct tap and hold functions.  One way to do it is to leverage the existing support for `MT()` and `LT()`: see the [Intercepting Mod-Taps](https://docs.qmk.fm/#/mod_tap?id=intercepting-mod-taps) section in the manual for some examples (in particular, [Changing hold function](https://docs.qmk.fm/#/mod_tap?id=changing-hold-function`)).

Lorn888, 2023-04-15:
Thank you for the reply. Sorry I didn't use discord, I don't know why. What i'm trying to do is press and hold the cmd button and than type "G" "I". And in return I want to trigger a  (Ctrl+Shift+1) combo.

andrebrait, 2023-05-01:
Also, from your code, I'm not sure the "hold" will work for your G key.

Your matrix code will set g_pressed to false after the time expires and so your code for the G being released won't do anything. And even then, it would only do stuff when you released the G key after that amount of time. If you kept pressing it, it would do nothing (even if the matrix scan didn't set g_pressed to false, that is).

I think you want to set g_pressed variable to true/false on your process records function, and send the keypresses inside the matrix scan funciona (which will execute every time, regardless of new keys getting pressed or not, so you can check the timer there).

Otherwise, you'll press it, set g_pressed to true, then the time will elapse, the matrix scan function will set it to false, and the handling of the release of the G key won't do anything.

closed 2023-08-31 by github-actions[bot]

PeterMortensen, 2025-03-08:
Note: `KC_SNAP` is [a Keychron-specific custom keycode](https://www.reddit.com/r/Keychron/comments/1j61n3b/comment/mgol0mg/) (Mac-specific).