qmk/qmk_firmware #20459: Keychron K3 Pro RGB effect issues

MarcoBuess, 2023-04-15:
### Issue Description

I recently got my first QMK board (Keychron K3 Pro) and am experiencing a weird issue with some RGB effects. Affected effects are: 

1. REACTIVE_SIMPLE
2. REACTIVE_MULTIWIDE
3. REACTIVE_MULTINEXUS

The effects work for literally every key but the `-`Key (MINUS). I got a `ISO-DE` board if that matters in any way.

I'm pretty certain this is a software related issue as the switch itself and the lighting on the switch is working for other effects just fine (e.g. static lighting). I ran a key test through VIA which showed the key is working correctly. I also did a factory reset and flashed the latest firmware as described in [K3 Pro Factory Reset and Firmware Flash (keychron.com)](https://www.keychron.com/blogs/archived/k3-pro-factory-reset-and-firmware-flash).

I'm fighting myself through the QMK docs for a couple of days now and am fairly certain that if someone could point me in the right direction I might be able to fix the issue and recompile the firmware myself. 

Any help would be highly appreciated. ❤

MarcoBuess, 2023-04-16:
I guess I found something sus in [rgb.c](https://github.com/Keychron/qmk_firmware/blob/bluetooth_playground/keyboards/keychron/k3_pro/iso/rgb/rgb.c)

That `NO_LED` thing between `70` and `71` could this be the cause of my issue? That is right around where the `-` key is located on the layout.

```
led_config_t g_led_config = {
    {
        { 0,      1,      2,      3,      4,      5,      6,      7,      8,      9,      10,     11,     12,     13,     14,     15 },
        { 16,     17,     18,     19,     20,     21,     22,     23,     24,     25,     26,     27,     28,     29,     NO_LED, 30 },
        { 31,     32,     33,     34,     35,     36,     37,     38,     39,     40,     41,     42,     43,     44,     NO_LED, 45 },
        { 46,     47,     48,     49,     50,     51,     52,     53,     54,     55,     56,     57,     NO_LED, 58,     NO_LED, 59 },
        { 60,     61,     62,     63,     64,     65,     66,     67,     68,     69,     70,     NO_LED, 71,     72,     73,     74 },
        { 75,     76,     77,     NO_LED, NO_LED, NO_LED, 78,     NO_LED, NO_LED, NO_LED, 79,     80,     81,     82,     83,     84 }
    },
```

lalalademaxiya1, 2023-06-20:
@MarcoBuess Hi, thank you for your interest in Keychron. The issue has been fixed, and the new firmware has been updated on the official website as well. Would you please close this PR at your convenience? Thank you.

MarcoBuess, 2023-06-20:
Sure no problem. Thanks for the support 👍

closed 2023-06-20 by MarcoBuess