qmk/qmk_firmware #20493: [Feature Request] Digital joystick acting as a switch

seflm, 2023-04-18:
### Feature Request Type

- [ ] Core functionality
- [X] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [X] New behavior

### Description

Hello,

I've got a QVEX Lynepad macro keypad that has an RKJXT1F42001 digital joystick (or rather 4-way soft/hard push switch with rotation and push functionality). I'd like to make this work with QMK and VIA. However, there seems to be no way of implementing this right now. There is the joystick feature in QMK but that is designed for gaming purposes. In this case, the joystick is supposed to be pressing macros/shortcuts or other actions when tilted.

The rotation and push functionality can be mapped using a regular encoder functionality of the QMK. But I see no way of implementing the soft/hard tilt into the QMK in a way it can be setup using the QMK configurator or VIA.

elpekenin, 2023-04-19:
[image]
Im no hardware expert but this looks like it's electrically: 4 switches for the directions + another for the press + an encoder, so it wouldnt need extra code to work... And unless the hardware has been changed since, somebody already ported it to QMK a couple years ago and the code should work #10826

sigprof, 2023-04-19:
Honestly, the implementation of the encoder switches (including the simple push one on the second encoder) in #10826 is rather bad — it's not integrated into the switch matrix as it should be (unfortunately, QMK makes combining the key matrix with some extra direct pins rather difficult — you basically need to implement a custom matrix, even though most of the code could just be copied from the default matrix implementation).

And specifically for RKJXT1F42001 there is an additional difficulty due to how it works — the directional press is intended to be detected when the center push switch clicks (the A/B/C/D switches close earlier without any tactile feedback).  So if you just map the Push, A, B, C, D switches to some key matrix locations, you may see some accidental activations of the A/B/C/D switches alone (if the sideways force was not enough to close the Push switch), and proper directional presses would appear as both the Push and one of A/B/C/D switches activating at the same time.  You could try to use the Combo feature to detect such actions, but it might not work if the delay between the activation of the A/B/C/D switch and the Push switch would be larger than expected.  So maybe in this particular case the proper way would be to preprocess the directional switch inputs in the custom matrix code.

seflm, 2023-05-12:
Thank you for the suggestions. I managed to write a custom matrix that includes the directional switches. To make it VIA compatible, I had to make two "virtual" matrix rows (not sure, if that was a good call).

closed 2023-05-12 by seflm