qmk/qmk_firmware #20535: CATERINA DISCONNECTING ALWAYS AFTER 4 SECONDS

Shaytan05, 2023-04-23:
### Issue Description

Hello guys, I'm about to flash the firmware on my Pro Micro ATmega32U4. I plug the SUB, reset twice and it connect to QMK Toolbox enabling flash button, but after 3 or 4 seconds it disconnect. I already tried different USB port and cable, but nothing change. 

https://ibb.co/DYpY1wT

What am I missing?

sigprof, 2023-04-23:
This is the expected behavior of the Caterina bootloader (actually the reference implementation should have a 8 second timeout, but maybe some parts of that time get eaten by Windows doing its things).  You need to press the “Flash” button before the timeout expires, or enable the “Auto-Flash” checkbox (unfortunately, in some configurations even that could be unreliable).

Shaytan05, 2023-04-23:
> This is the expected behavior of the Caterina bootloader (actually the reference implementation should have a 8 second timeout, but maybe some parts of that time get eaten by Windows doing its [...]

I was "scared" to tap the flash button, but it worked! Thanks mate.

I'm building a manuform redox keyboard, should I flash the same firmware also on the other Board? Can I use them separately then?

Thanks in advice

sigprof, 2023-04-23:
Most QMK split boards use the same firmware for both sides (except some really weird ones).

If your board uses the `redox/rev1/base` firmware, it is configured with `MASTER_LEFT` by default, so you must plug USB to the left side to make it behave properly; plugging USB into the right side would make it work as a mirrored left side.  Fixing that would require changing the handedness detection method — `EE_HANDS` should work without any hardware modifications, but you would need to flash the handedness information into the EEPROM once (QMK Toolbox has commands for that in the menu).  If you do that, you should be able to use both halves of the board separately, with each half behaving as the appropriate side (but any layer switching won't affect the other half, even if it is connected to the same computer over USB).

Shaytan05, 2023-04-26:
> Most QMK split boards use the same firmware for both sides (except some really weird ones).
>
> If your board uses the `redox/rev1/base` firmware, it is configured with `MASTER_LEFT` by default, so you [...]

I'm very newbie with this, I didn't understand very well what I should do.
At the moment I completed also the right hand, but if I connect it to the left with TRRS connector (I didn't use the resistors, can be that the problem??) nothing happens. I flashed the firmware also on the right hand to check if I built it correctly, and if I plug it is a mirrorred left hand (as you said, but this prove that it is correctly wired).

I can't undestand why they don't works together.....

Can you help me please?

myst729, 2023-05-09:
> I'm very newbie with this, I didn't understand very well what I should do. At the moment I completed also the right hand, but if I connect it to the left with TRRS connector (I didn't use the [...]

Most code could be reused on both halves. But you need to specify which is which. That means you need to compile two copies of firmware, with a few lines of different config. One for left hand and the other for right hand. https://docs.qmk.fm/#/feature_split_keyboard

myst729, 2023-05-09:
Take a look at the docs https://docs.qmk.fm/#/feature_split_keyboard

Shaytan05, 2023-05-21:
> Take a look at the docs https://docs.qmk.fm/#/feature_split_keyboard

I have a lot of confusion now... in the docs you linked to me there's the picture that shows hot to wire the Arduino pro micro in serial connection, but seems it doens't fit with this: https://github.com/mattdibi/redox-keyboard/blob/master/redox/README.md#assembly-right-hand-as-master.

Now I really don't know how to wire my TRRS connector. Now it's wired this way:
TRRS LEFT SIDE: GND-PD1-PD0-D0(ON THE STRIP)
TRRS RIGHT SIDE: GND-PD1-PD0-D0(ON THE STRIP)
But this seems to be the I2C connection... I don't use resistors as they say I don't need them with led strips.

closed 2023-09-19 by github-actions[bot]