qmk/qmk_firmware #20536: Not blank at 0x1 after flash

jgrubio, 2023-04-23:
### Issue Description

Hi,

Yesterday I compiled without problems: [markstos](https://github.com/qmk/qmk_firmware/tree/master/keyboards/crkbd/keymaps/markstos) ( I did not see that it had errors ).

Copy the 3 files in my pc and when doing the flash it appeared: **Checking memory from 0x0 to 0x6FFF... Not blank at 0x1**.

Complete log:

```
qmk flash -kb crkbd -km markstos
Ψ Compiling keymap with make --jobs=1 crkbd/rev1:markstos:flash

QMK Firmware 0.20.5
Making crkbd/rev1 with keymap markstos and target flash

avr-gcc (GCC) 12.2.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text	   data	    bss	    dec	    hex	filename
      0	  18252	      0	  18252	   474c	crkbd_rev1_markstos.hex

Copying crkbd_rev1_markstos.hex to qmk_firmware folder                                              [OK]
Checking file size of crkbd_rev1_markstos.hex                                                       [OK]
 * The firmware size is fine - 18252/28672 (63%, 10420 bytes free)
Flashing for bootloader: atmel-dfu
Bootloader Version: 0x00 (0)
Checking memory from 0x0 to 0x6FFF...  Not blank at 0x1.
Erasing flash...  Success
Checking memory from 0x0 to 0x6FFF...  Empty.
Checking memory from 0x0 to 0x477F...  Empty.
0%                            100%  Programming 0x4780 bytes...
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
0%                            100%  Reading 0x7000 bytes...
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
Validating...  Success
0x4780 bytes written into 0x7000 bytes memory (63.84%).
make[1]: *** [platforms/avr/flash.mk:173: flash] Error 254
Make finished with errors
make: *** [Makefile:392: crkbd/rev1:markstos:flash] Error 1
```
Now every time I flash any configuration I get the same error.

I have tried to launch:

```
dfu-programmer atmega32u4 erase --force
Erasing flash... Success
Checking memory from 0x0 to 0x6FFF... Empty
```

```
dfu-programmer atmega32u4 flash --force .build/crkbd_rev1_default.hex
0% 100% Programming 0x6900 bytes...
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>] Success
0% 100% Reading 0x7000 bytes...
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>] Success
Validating... Success
0x6900 bytes written into 0x7000 bytes memory (93.75%).
```

```
dfu-programmer atmega32u4 reset
```

And although it seems that everything is fine, when I then flash using qmk as I explain above, it puts me again the error.

**SO:** Arch linux
**qmk version:** 1.1.2
**Keyboard:** Helidox (Corne) R2G

sigprof, 2023-04-23:
I suppose that if you run `dfu-programmer atmega32u4 erase` (without `--force`), it would print:
```
Checking memory from 0x0 to 0x6FFF...  Not blank at 0x1.
Erasing flash...  Success
Checking memory from 0x0 to 0x6FFF...  Empty.
```
So in this case the `Not blank at 0x1` refers to the initial state before erase (the `erase` command without `--force` checks whether the flash is already empty to avoid an unneeded erase, but apparently emits a dangerously looking message because of that).  If you see the “Empty” report after “Erasing flash”, you should ignore the previous “Not blank at 0x1” report.

As for “Error 254”, not sure where does it come from — apparently the exit code of `dfu-programmer` is not 100% reliable in some configurations, so you may need to rely on eyeballing “Validating...  Success” in the output and ignoring the spurious error in that case.  I suppose that the keyboard actually works fine after that “Error 254”?

But the real problem is that apparenty Arch Linux updated their `dfu-programmer` package to version 1.0.0, and the flashing code in QMK does not detect that version properly and tries to use commands intended for the ancient 0.6.x versions, which did not have the `--force` option.  This would need to be fixed somehow…

jgrubio, 2023-04-23:
> I suppose that if you run `dfu-programmer atmega32u4 erase` (without `--force`), it would print:
> 
> ```
> Checking memory from 0x0 to 0x6FFF...  Not blank at 0x1.
> Erasing flash...  Success
> Checking memory from 0x0 to 0x6FFF...  Empty.
> ```

Exactly, if i run `dfu-programmer atmega32u4 erase` i get the same output.

> So in this case the `Not blank at 0x1` refers to the initial state before erase (the `erase` command without `--force` checks whether the flash is already empty to avoid an unneeded erase, but [...]

> As for “Error 254”, not sure where does it come from — apparently the exit code of `dfu-programmer` is not 100% reliable in some configurations, so you may need to rely on eyeballing “Validating... [...]

Understood. :heavy_check_mark: 
Yes, apparently the keyboard actually works fine.

> But the real problem is that apparenty Arch Linux updated their `dfu-programmer` package to version 1.0.0, and the flashing code in QMK does not detect that version properly and tries to use [...]

I confirm that I have the version also:

`dfu-programmer --version
dfu-programmer 1.0.0
`

It seems strange because  yesterday with the same versions of everything (OS, qmk etc...) I didn't get the error 254 (flashing 10 times maybe) and after flashing using the _markstos_ configuration it always appears. I say this, in case it can help you in debugging the problem.

Thank you very much for the help. I will try in another SO like Ubuntu and I will post the behavior.

sigprof, 2023-04-23:
That 254 is likely the result of truncating this `-2` to `unsigned char`:
https://github.com/dfu-programmer/dfu-programmer/blob/v1.0.0/src/atmel.c#L901

You may be able to see the debug messages if you run `dfu-programmer atmega32u4 reset --debug 50`.

The “reset” command should eventually cause USB disconnect when the bootloader exits and reboots into the actual firmware; I suspect that on some machines this disconnect may happen before the host actually receives the reply for the reset command, and that `dfu-programmer` code may detect this as an error, even though things are actually fine.

jgrubio, 2023-04-25:
I let you here the logs when I launch this with `dfu-programme`.

```
dfu-programmer atmega32u4 erase --force 
Erasing flash...  Success
Checking memory from 0x0 to 0x6FFF...  Empty.
```

```
dfu-programmer atmega32u4 flash --force .build/crkbd_rev1_default.hex 
0%                            100%  Programming 0x6900 bytes...
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
0%                            100%  Reading 0x7000 bytes...
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
Validating...  Success
0x6900 bytes written into 0x7000 bytes memory (93.75%).
```

```
dfu-programmer atmega32u4 reset --debug 50
     target: atmega32u4
    chip_id: 0x2ff4
  vendor_id: 0x03eb
    command: reset
      quiet: false
      debug: 50
device_type: AVR
------ command specific below ------
```
I have not yet had the opportunity to test it in another Operating System other than Ach and Windows.

closed 2023-09-07 by github-actions[bot]