qmk/qmk_firmware #20537: [Bug] Q6 FN Key not working

Fischje, 2023-04-23:
### Describe the Bug

I updated my Firmware of my Q6 iso knob Version to 1.5 from keychron website.

In Via-App i can see FN key pressed in matrix mode, but in normal mode the layout is the us layout not iso and my FN key is not woring.
Is it a firmware issue or an issue with via-app.

### Keyboard Used

Keychron Q6 Iso Knowb

### Operating System

Win10

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

ChaoticAlex, 2023-04-24:
QMK does not use pure FN keys, but rather more advanced and diverse layer management, see https://docs.qmk.fm/#/feature_layers, so it's common to FN key is not woring in VIA normal mode.
Maybe you can confirm the keycode and try to see if the actual function is working.

drashna, 2023-05-22:
Specifically, the "fn" keys are internal layer keys.  These are not exposed to the OS/host at all, and are solely used internally.

VIA's keylogger works because it sends the column and row matrix information to the host system via VIA, and then renders it.  So it's not actually looking at keypresses, but rather is looking at matrix activation.

lalalademaxiya1, 2023-06-20:
@Fischje Hi, Do you have any further quentions? If everything is fine, would you please close this PR at your convenience? Thanks a lot.

closed 2023-11-01 by zvecr