qmk/qmk_firmware #20668: [Bug] mod-tap on combo toggles mod into key-down state

filterpaper, 2023-05-02:
### Describe the Bug

PR #15847 appears to have triggered a regression with using mod-tap keys in a combo. Mod is activated and deactivated correctly when combo held down longer than tapping term. When the combo is tapped within tapping term, modifier key-down state is sent.

Test case:
```c
enum combos {
  modtest,
  COMBO_LENGTH
}

uint16_t COMBO_LEN = COMBO_LENGTH;
uint16_t const modtest_combo[] PROGMEM = {KC_Y, KC_U, COMBO_END};

combo_t key_combos[] = {
  [modtest] = COMBO(modtest_combo, RSFT_T(KC_SPC))
}
```

- `KC_Y` + `YC_U` held down longer than `TAPPING_TERM`: Right-Shift key-down is sent.
- `KC_Y` + `YC_U` released after `TAPPING_TERM`: Right-Shift key-up is sent.
- `KC_Y` + `YC_U` tapped and released under `TAPPING_TERM`: Right-Shift key-up followed by Right-Shift key-down event is sent to host.

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

sigprof, 2023-05-02:
#15847 did this in the combo press code:
```diff
@@ -332,8 +332,8 @@ void apply_combo(uint16_t combo_index, combo_t *combo) {
         KEY_STATE_DOWN(state, key_index);
         if (ALL_COMBO_KEYS_ARE_DOWN(state, key_count)) {
             // this in the end executes the combo when the key_buffer is dumped.
-            record->keycode   = combo->keycode;
-            record->event.key = MAKE_KEYPOS(KEYLOC_COMBO, KEYLOC_COMBO);
+            record->keycode    = combo->keycode;
+            record->event.type = COMBO_EVENT;
 
             qrecord->combo_index = combo_index;
             ACTIVATE_COMBO(combo);
```
Looks like the `record->event.key` assignment which was removed here still needs to be made (or maybe `record->event = MAKE_COMBOEVENT(true);` should be used instead of patching the `event` fields separately, so that it always matches the combo release event).  The tapping code uses `KEYEQ(tapping_key.event.key, (r->event.key))` even for `COMBO_EVENT` (it rejects `TICK_EVENT` and then treats all other kinds of events the same), so a `record->event.key` mismatch between press and release events would break it.

sevanteri, 2023-05-02:
Just tested adding `record->event  = MAKE_COMBOEVENT(true);` there and it indeed seems to fix this.

filterpaper, 2023-05-08:
The regression has also impacted combos with OSM. Test case:
```c
enum combos {
  modtest,
  osmshift,
  COMBO_LENGTH
}

uint16_t COMBO_LEN = COMBO_LENGTH;
uint16_t const modtest_combo[] PROGMEM = {KC_Y, KC_U, COMBO_END};
uint16_t const osmshift_combo[] PROGMEM = {KC_Z, KC_X, COMBO_END};

combo_t key_combos[] = {
  [modtest] = COMBO(modtest_combo, RSFT_T(KC_SPC)),
  [osmshift] = COMBO(osmshift_combo, OSM(MOD_LSFT))
}
```
Regressed behaviour:
* `KC_Z`+`KC_X` tapped and released: Left-Shift key-down is sent to host and remains down

Expected behaviour:
* Tap `KC_Z`+`KC_X` tapped and released: No report to host
* Tap `KC_I` tapped and released: 
  * Left-Shift key-down
  * `KC_I` key-down
  * Left-Shift key-up
  * `KC_I` key-up

PR #20669 (2023-05-10): Fix Mod-Tap combo regression
## Description

Fixes #20668 by adding the keyevent back to the keyrecord.

## Types of Changes

- [ ] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* #20668 

## Checklist

- [x] My code follows the code style of this project: [**C**](https://docs.qmk.fm/#/coding_conventions_c), [**Python**](https://docs.qmk.fm/#/coding_conventions_python)
- [x] I have read the [**PR Checklist** document](https://doc [... cut]
closed 2023-05-10 by filterpaper