qmk/qmk_firmware #20673: Incorrect/broken image links

jepler, 2023-05-02:
### Issue Description

While checking README files, I noticed that some imgur URLs are broken or of the wrong form to link directly to an image (imgur.com/a/ urls). This means that when viewing the markdown file, the intended image is not shown. I spot checked a few by looking in the github markdown viewer and it seems that my script's determination was correct, at least in most cases.

The raw URLs:
```
https://imgur.com/a/5AwQw0r
https://imgur.com/a/5v5pV9U
https://imgur.com/gallery/tPDHeB9?s=fbm
https://imgur.com/Gyim08a
https://imgur.com/a/riUxwpl
https://imgur.com/hCDdQIc
https://i.imgur.com/T5Lm6OYlh.png
https://i.imgur.com/
https://imgur.com/uYCeS4Y
https://imgur.com/a/xIpn2U5
https://i.imgur.com/midAJSplh.jpeg
https://imgur.com/a/cJpZQBn
https://imgur.com/H2Yqnv3
http://imgur.com/aAi6lDe
https://imgur.com/gallery/Pwwb0Q0
https://imgur.com/a/P6eSrcy
https://imgur.com/a/Tw7dwK4
https://imgur.com/SF4UP2j
https://imgur.com/1chZff9
http://imgur.com/ZWIfuPM
https://imgur.com/X1c1qeY
```

The markdown files where they appear:
```markdown
keyboards/takashiski/hecomi/alpha/readme.md:![hecomi_alpha](https://imgur.com/a/5AwQw0r)
keyboards/takashiski/hecomi/readme.md:![hecomi](https://imgur.com/a/5AwQw0r)
keyboards/exclusive/e7v1/readme.md:![E7-V1](https://imgur.com/a/5v5pV9U)
keyboards/snampad/readme.md:![snampad](https://imgur.com/gallery/tPDHeB9?s=fbm)
keyboards/gray_studio/think65/solder/keymaps/dangjoeltang/readme.md:![Base Layer](https://imgur.com/Gyim08a)
keyboards/maple_computing/lets_split_eh/keymaps/resfury/readme.md:![Let's Split Eh?](https://imgur.com/a/riUxwpl)
keyboards/tweetydabird/lbs4/keymaps/default/readme.md:![lsb4](https://imgur.com/hCDdQIc)
keyboards/mechlovin/mechlovin9/rev3/readme.md:![mechlovin9](https://i.imgur.com/T5Lm6OYlh.png)
keyboards/viktus/styrka/readme.md:![Styrka](https://i.imgur.com/)
keyboards/molecule/readme.md:[![molecule](https://imgur.com/uYCeS4Y)](https://i.imgur.com/uYCeS4Yh.jpg)
keyboards/primekb/prime_o/keymaps/reasonsandreasons/readme.md:![reasonsandreasons Layout Image](https://imgur.com/a/xIpn2U5)
keyboards/mechlovin/olly/orion/readme.md:![Olly Orion](https://i.imgur.com/midAJSplh.jpeg)
keyboards/exclusive/e65/keymaps/masterzen/readme.md:![masterzen keymap Layout Image](https://imgur.com/a/cJpZQBn)
keyboards/gray_studio/think65/solder/keymaps/dangjoeltang/readme.md:![Function Layer](https://imgur.com/H2Yqnv3)
keyboards/kc60/keymaps/stanleylai/readme.md:![Base Layer](http://imgur.com/aAi6lDe)
keyboards/qwertyydox/readme.md:![QWERTYYdox](https://imgur.com/gallery/Pwwb0Q0)
keyboards/yeehaw/readme.md:![Yeehaw](https://imgur.com/a/P6eSrcy)
keyboards/handwired/oem_ansi_fullsize/readme.md:![oem_ansi_fullsize](https://imgur.com/a/Tw7dwK4)
keyboards/exclusive/e7v1/keymaps/masterzen/readme.md:![masterzen keymap Layout Image](https://imgur.com/SF4UP2j)
keyboards/rgbkb/sol/readme.md:![Sol](https://imgur.com/1chZff9)
keyboards/kc60/keymaps/stanleylai/readme.md:![RGB Layer](http://imgur.com/ZWIfuPM)
keyboards/capsunlocked/cu24/keymaps/default/readme.md:![CU24 Layout Image](https://imgur.com/X1c1qeY)
```

FLinguenheld, 2023-05-25:
Hello,
For me it works with the final image's link.
For instance with your first image which is https://imgur.com/a/5AwQw0r :

[image] ✗
[image] ✓

jepler, 2023-08-28:
Hey, this is just to let you know that _I_ do not plan to take any specific action on this issue.

closed 2023-12-29 by github-actions[bot]