qmk/qmk_firmware #20729: [Bug] SPLIT_HAPTIC_ENABLE fix accidentally removed again?

kilipan, 2023-05-05:
### Describe the Bug

When compiling a split keyboard with haptic module only on the primary side, i.e. `HAPTIC_ENABLE` is defined, but `SPLIT_HAPTIC_ENABLE` isn't, the following error is thrown:
```C
Compiling: quantum/matrix_common.c                                                                 In file included from quantum/split_common/transactions.h:24,
                 from quantum/matrix_common.c:9:
quantum/split_common/transport.h:204:5: error: unknown type name 'split_slave_haptic_sync_t'
  204 |     split_slave_haptic_sync_t haptic_sync;
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~
```

I believe this is a consequence of the fix in #20411 being reverted for some reason. See <https://github.com/qmk/qmk_firmware/blob/master/quantum/split_common/transport.h#L203-L205>, which reads
```C
#if defined(HAPTIC_ENABLE)
    split_slave_haptic_sync_t haptic_sync;
#endif // defined(HAPTIC_ENABLE)
```
If I'm not mistaken, it should be
```C
#if defined(HAPTIC_ENABLE) && defined(SPLIT_HAPTIC_ENABLE)
    split_slave_haptic_sync_t haptic_sync;
#endif // defined(HAPTIC_ENABLE) && defined(SPLIT_HAPTIC_ENABLE)
```

### Keyboard Used

[custom]

### Operating System

Void Linux

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /home/js/qmk_firmware
Ψ Detected Linux (Void Linux).
Ψ Git branch: master
Ψ Repo version: 0.20.6
Ψ - Latest master: 2023-05-02 05:28:05 +0200 (9b41e8168d) -- [Keyboard] add phantagom/baragon marcopad (#20543)
Ψ - Latest upstream/master: 2023-05-04 15:29:45 -0400 (7f413e9f01) -- [Keyboard] Add 1upkeyboards pi50 (#20242)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2023-05-02 05:28:05 +0200 (9b41e8168d) -- [Keyboard] add phantagom/baragon marcopad (#20543)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 9.3.0
Ψ Found avr-gcc version 9.3.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 7.0
Ψ Found dfu-programmer version 0.7.2
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-01-03 19:29:26 +0000 --  (0062927e3)
Ψ - lib/chibios-contrib: 2023-01-11 16:42:27 +0100 --  (a224be15)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Other keyboard-related software installed

none

sigprof, 2023-05-05:
The fix is definitely present in the code, but currently only in the `develop` branch.  It will be available in the `master` branch after the next round of [breaking changes](https://docs.qmk.fm/#/breaking_changes), scheduled for May 28, 2023.

kilipan, 2023-05-05:
Oh, I see, great! And sorry then! I'm not very familiar with the codebase, was just looking for a quick fix to my problem.
I assume I can close the issue then :blush:

closed 2023-05-05 by kilipan