qmk/qmk_firmware #20772: Keyboard disconnecting when changing RGB mode.

gabxln, 2023-05-08:
### Describe the Bug

When I'm switching RGB modes, after reaching the last effect the keyboard disconnect, freeze, from the computer and I need to plug my keyboard again. This only happens when I'm changing the RGB effect. This doesn't happen when I use VIA or other command from QMK like speed or brightness.

[image]

### Keyboard Used

kprepublic/bm60hsrgb/rev2:default

### Link to product page (if applicable)

https://kprepublic.com/products/bm60-rgb-60-gh60-hot-swappable-pcb-programmed-qmk-firmware-type-c

### Operating System

Windows 10 22H2 19045.2913

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Other keyboard-related software installed

VIA

### Additional Context

QMK 0.2.2
QMK Configurator 

[image]

drashna, 2023-05-22:
If you lower the brightness (to as low as possible, ideally), does this still happen? 

Also, to any more, or a specific mode?   Because it sounds like it *may* be pulling too much power?

gabxln, 2023-05-22:
> If you lower the brightness (to as low as possible, ideally), does this still happen?
> 
> Also, to any more, or a specific mode? Because it sounds like it _may_ be pulling too much power?

no, only with the RGB effect.