qmk/qmk_firmware #20963: [Bug] Slightly different timing with M2 Mac

richtera, 2023-05-15:
### Describe the Bug

I am using my cursor keys by having left and right and then using left as a layer switch. This has worked on Windows, Ubuntu and Mac, but recently I got a M2 MBP and about 1/2 of the time it doesn't see my layer switch in time and I have to hit the sequence twice before it works. I am not quite sure where to start looking what needs to be done to get the timing right. The same identical configuration worked on my MBP Intel for a really long time, so I think it's something arm specific.

### Keyboard Used

Planck v6

### Operating System

Mac OS Ventura 13.3.1 (a) (22E772610a)

### qmk doctor Output

```text
I used the online builder with the QMK configurator
Here is my config file

{"version":1,"notes":"","documentation":"\"This file is a QMK Configurator export. You can import this at <https://config.qmk.fm>. It can also be used directly with QMK's source code.\n\nTo setup your QMK environment check out the tutorial: <https://docs.qmk.fm/#/newbs>\n\nYou can convert this file to a keymap.c using this command: `qmk json2c {keymap}`\n\nYou can compile this keymap using this command: `qmk compile {keymap}`\"\n","keyboard":"planck/rev6","keymap":"planck_rev6_layout_ortho_4x12_mine","layout":"LAYOUT_ortho_4x12","layers":[["KC_TAB","KC_Q","KC_W","KC_E","KC_R","KC_T","KC_Y","KC_U","KC_I","KC_O","KC_P","KC_BSPC","KC_ESC","KC_A","KC_S","KC_D","KC_F","KC_G","KC_H","KC_J","KC_K","KC_L","KC_SCLN","KC_QUOT","KC_LSFT","KC_Z","KC_X","KC_C","KC_V","KC_B","KC_N","KC_M","KC_COMM","KC_DOT","KC_SLSH","KC_RSFT","BL_STEP","KC_LCTL","KC_LALT","KC_LGUI","MO(3)","KC_SPC","KC_SPC","MO(4)","LT(1,KC_LEFT)","LT(2,KC_RGHT)","KC_RALT","KC_ENT"],["KC_NO","RGB_VAI","RGB_SPI","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_BSPC","KC_NO","RGB_VAD","RGB_SPD","BL_ON","BL_INC","LCG_SWP","KC_NO","KC_HOME","KC_UP","KC_END","KC_NO","KC_DEL","KC_LSFT","BL_TOGG","BL_STEP","BL_OFF","BL_DEC","LCG_NRM","KC_PGUP","KC_LEFT","KC_DOWN","KC_RGHT","KC_PGDN","KC_RSFT","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO"],["KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_MS_U","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_MS_L","KC_MS_D","KC_MS_R","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_BTN1","KC_NO","KC_BTN2","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO"],["KC_TILD","KC_EXLM","KC_AT","KC_HASH","KC_DLR","KC_PERC","KC_CIRC","KC_AMPR","KC_ASTR","KC_LPRN","KC_RPRN","KC_BSPC","KC_DEL","KC_F1","KC_F2","KC_F3","KC_F4","KC_F5","KC_F6","KC_UNDS","KC_PLUS","KC_LCBR","KC_RCBR","KC_PIPE","KC_TRNS","KC_F7","KC_F8","KC_F9","KC_F10","KC_F11","KC_F12","LSFT(KC_NUHS)","LSFT(KC_NUBS)","KC_HOME","KC_END","KC_TRNS","KC_PSCR","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","MO(6)","KC_MNXT","KC_VOLD","KC_VOLU","KC_MPLY"],["KC_GRV","KC_1","KC_2","KC_3","KC_4","KC_5","KC_6","KC_7","KC_8","KC_9","KC_0","KC_BSPC","KC_DEL","KC_F1","KC_F2","KC_F3","KC_F4","KC_F5","KC_F6","KC_MINS","KC_EQL","KC_LBRC","KC_RBRC","KC_BSLS","KC_TRNS","KC_F7","KC_F8","KC_F9","KC_F10","KC_F11","KC_F12","KC_NUHS","KC_NUBS","KC_PGUP","KC_PGDN","KC_TRNS","KC_PSCR","KC_TRNS","KC_TRNS","KC_TRNS","MO(6)","KC_TRNS","KC_TRNS","KC_TRNS","KC_MNXT","KC_VOLD","KC_VOLU","KC_MPLY"],["KC_1","KC_1","KC_1","KC_1","KC_1","KC_1","KC_1","KC_1","KC_1","KC_1","KC_1","KC_1","KC_NO","KC_Q","KC_W","KC_E","KC_R","KC_T","KC_Y","KC_U","KC_I","KC_O","KC_P","KC_LBRC","KC_NO","KC_A","KC_S","KC_D","KC_F","KC_G","KC_H","KC_J","KC_K","KC_L","KC_SCLN","KC_QUOT","TO(0)","KC_NO","KC_NO","KC_C","KC_V","KC_NO","KC_NO","KC_N","KC_M","KC_NO","KC_NO","KC_NO"],["KC_TRNS","RESET","DEBUG","RGB_TOG","RGB_MOD","RGB_HUI","RGB_HUD","RGB_SAI","RGB_SAD","RGB_VAI","RGB_VAD","KC_DEL","KC_TRNS","KC_TRNS","ANY(MU_MOD)","ANY(AU_ON)","ANY(AU_OFF)","AG_NORM","AG_SWAP","DF(0)","KC_NO","KC_NO","TO(5)","KC_TRNS","KC_TRNS","ANY(MUV_DE)","ANY(MUV_IN)","ANY(MU_ON)","ANY(MU_OFF)","ANY(MI_ON)","ANY(MI_OFF)","ANY(TERM_ON)","ANY(TERM_OFF)","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS"]],"author":""}
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [X] Karabiner (macOS)

drashna, 2023-05-22:
Does the same board and firmware work on the other systems fine?  Eg, is this issue local ONLY to the M2 mac?

richtera, 2023-05-22:
Yes, I use the same board with the same firmware on my Ubuntu laptop, Ubuntu nuk, Mac Intel, Windows and Raspberry pi. Only the Mac M2 has the problem as far as I can detect. I constantly have to re-type the Cursor up key which is on a temporary layer using my left cursor key as a layer key.

drashna, 2023-05-22:
Okay. And that's pretty odd.  I'd say that this is likely due to the M2's USB stack, if anything.

A simple solution may be to decrease the tapping term, and see if that helps.  That or play with some of the tap-hold config settings, as that may help too.