qmk/qmk_firmware #20972: [help] cannot use OLED with STM32F103RCT6.

KalamityStorm, 2023-05-17:
### Issue Description

hi,recently i want to build a keyboard for myself with STM32F103RCT6 core.
for some reason,i want my keyboard can use as more function as i can.
now i deployed MIDI keys/analog,OLED screen and Encorders,
but my OLED module seems error.

when i boot my code,system can't function normally my device. it said USB\VID_0000&PID_0002 (cannot reconize device),
i need to remove my pins to OLED SDA/SCL to boot keyboard normally.

is my code error?
or,where can i find a demo with STM32F103RCT6 to boot a OLED screen?
most i can find is for ATMEGA32U4,but seems these two uses different codes,and 32U4's io is too less to me.
thanks!

these are settings in my config.h;
`
#define I2C_DRIVER I2CD1
#define I2C1_CLOCK_SPEED 400000
#define I2C1_DUTY_CYCLE FAST_DUTY_CYCLE_2
#define I2C1_SCL_PIN B9
#define I2C1_SDA_PIN B8

#define OLED_BRIGHTNESS 255
#define OLED_TIMEOUT 80000 
#define OLED_FADE_OUT 
#define OLED_FADE_OUT_INTERVAL 5 
#define OLED_SCROLL_TIMEOUT 40000 

#define OLED_DISPLAY_128X64
`
these are settings in my mcuconf.h;
`#pragma once

#undef HAL_USE_I2C
#define HAL_USE_I2C TRUE

#undef HAL_USE_ADC
#define HAL_USE_ADC TRUE

#include_next <halconf.h>`

these are settings in my halconf.h;
`#pragma once

#include_next <mcuconf.h>

#undef STM32_I2C_USE_I2C1
#define STM32_I2C_USE_I2C1 TRUE

#define HAL_USE_I2C TRUE

#undef STM32_ADC_USE_ADC1
#define STM32_ADC_USE_ADC1 TRUE
`

glad for anyhelp!

sigprof, 2023-05-17:
> #define I2C1_SCL_PIN B9
> #define I2C1_SDA_PIN B8

These pin assignments are wrong — SDA must be B9, and SCL must be B8.  The functions of these pins cannot be swapped in software, so if your board is actually wired as SCL=B9, SDA=B8, you need to change the wiring physically.

Also STM32F103 is weird — changing the pin assignments for various peripherals needs to be done through some magic bits, which you need to set yourself (neither ChibiOS nor the QMK drivers handle that automatically).  So if you want to use SCL=B8, SDA=B9 for I2C1 instead of the default SCL=B6, SDA=B7, you would need to add the following piece of code:
```c
void board_init(void) {
    AFIO->MAPR |= AFIO_MAPR_I2C1_REMAP;
}
```

sigprof, 2023-05-17:
Also for STM32F103xC…xE you need to add a `board.h` file containing this:
```c
#pragma once

#include_next <board.h>

#undef STM32F103xB
#define STM32F103xE
```
Note that you need to specify `STM32F103xE` even if you actually have a xC chip — anything in the xC…xE range needs the `STM32F103xE` define.  Ideally you should also have `MCU_LDSCRIPT = STM32F103xC`, but that won't work without actually supplying the `STM32F103xC_stm32duino.ld` script (assuming that you are using the `stm32duino` bootloader); however, using the ld script for a smaller device should be safe.

The reason is that the STM32F103 family of chips actually consists of multiple subfamilies according to the flash size:
- low density — 16…32K (x4/x6);
- medium density — 64…128K (x8/xB);
- high density — 256…512K (xC/xD/xE);
- XL density — 768…1024K (xF/xG).

These subfamilies have some significant differences (even the set of peripherals is different); one important difference is that the high and XL density chips have a different flash sector size (2 KiB instead of 1 KiB), and this difference breaks the EEPROM emulation code if the firmware for a smaller chip is flashed into a larger chip.  The ChibiOS EFL driver determines the flash page size according to the chip type define (`STM32F103xB` or `STM32F103xE`), so redefining it is required to make things work correctly.

Also you probably need to add this to `config.h`, because the ChibiOS EFL driver for STM32F1xx does not have the code to read the flash size register:
```c
#define STM32_FLASH_SECTORS_PER_BANK 128
```
(128 sectors × 2 KiB = 256 KiB; the default is set up for a 512 KiB chip.)

Another test that could be useful here is adding `EEPROM_DRIVER = transient` to `rules.mk` — if that makes things work, there is some problem or misconfiguration in the flash driver.

KalamityStorm, 2023-05-17:
> 
thank for your help!i'll try it later.

KalamityStorm, 2023-05-18:
> These pin assignments are wrong — SDA must be B9, and SCL must be B8. The functions of these pins cannot be swapped in software, so if your board is actually wired as SCL=B9, SDA=B8, you need to [...]

hi,thankfully i've not build my pcb yet,so i tried to change SDA to B9,and SCL to B8,
this time i can boot normally,but the OLED doesn't show anything.
i'm wondering why,now i only leave these codes:

in config.h:

#define I2C1_SCL_PIN B8
#define I2C1_SDA_PIN B9
#define OLED_BRIGHTNESS 255
#define OLED_TIMEOUT 80000
#define OLED_FADE_OUT
#define OLED_FADE_OUT_INTERVAL 5 
#define OLED_SCROLL_TIMEOUT 40000 
#define OLED_DISPLAY_128X64

i didn;t change anything in mcuconf.h and chconf.h,and i make a new board.h as you said.
is there any settings i've forget?
or where can i find i demo in STM32F103 to use a oled screen?
thanks for your help!

sigprof, 2023-05-18:
Did you also add the `board_init()` function as above?

Also you may try using the default pins (SCL=B6, SDA=B7) in case there is something wrong with remapping these pins; in that case you don't need to add `board_init()`.

> or where can i find i demo in STM32F103 to use a oled screen?

There are some boards like that in the repo:
```
❯ qmk find -f processor=STM32F103 -f features.oled=true
Ψ Retrieving list of keyboards with keymap "default"...
Ψ Parsing data for all matching keyboard/keymap combinations...
Ψ Filtering on condition ("processor" == "STM32F103")...
Ψ Filtering on condition ("features.oled" == "true")...
doio/kb16/rev2:default
doio/kb30:default
handwired/pill60/bluepill:default
horrortroll/handwired_k552:default
rart/rartlice:default
rocketboard_16:default
```
None of them use the B8/B9 pins though (some of them use B10/B11 on I2C2 though).

Also, depending on which kind of display you got, you may need to add
```c
#define OLED_IC OLED_IC_SH1106
```
(and, if the above definition makes things somewhat work, maybe also `#define OLED_COLUMN_OFFSET 2` — the SH1106 chip supports 132 column pins, and the 128×64 screen may be connected to center columns, skipping 2 columns on each side).

KalamityStorm, 2023-05-18:
> 您是否也添加了`board_init()`上述功能？
> 
> 您也可以尝试使用默认引脚（SCL=B6，SDA=B7），以防重新映射这些引脚时出现问题；在这种情况下，您不需要添加`board_init()`.
> 
> > 或者我在哪里可以找到我在 STM32F103 中使用 OLED 屏幕的演示？
> 
> 回购中有一些这样的板：
> 
> ```
> ❯ qmk find -f processor=STM32F103 -f features.oled=true
> Ψ Retrieving list of keyboards with keymap "default"...
> Ψ Parsing data for all matching keyboard/keymap combinations...
> Ψ Filtering on condition ("processor" == "STM32F103")...
> Ψ Filtering on condition ("features.oled" == "true")...
> doio/kb16/rev2:default
> doio/kb30:default
> handwired/pill60/bluepill:default
> horrortroll/handwired_k552:default
> rart/rartlice:default
> rocketboard_16:default
> ```
> 
> 尽管它们都没有使用 B8/B9 引脚（尽管它们中的一些在 I2C2 上使用 B10/B11）。
> 
> 此外，根据您获得的显示器类型，您可能需要添加
> 
> ```c
> #define OLED_IC OLED_IC_SH1106
> ```
> 
> （而且，如果上面的定义使事情有点奏效，也许还有`#define OLED_COLUMN_OFFSET 2`——SH1106 芯片支持 132 列引脚，128×64 屏幕可以连接到中间列，每边跳过 2 列）。

really thanks for your help.now i completely know the reason.
first,i don't need to define like I2C_DRIVER and I2C_CLOCK_SPEED,i just only to define the IC module and resolution.
when i clear the error code,it's functionally as i want,thanks,i both check SH1106/SSD1306 or SSD1315 with 128*64 or 128*32,all good!

can i also ask a quesion about this,when i use SS1106(1.3 inch ver),except the define OLED_IC OLED_IC_SH1106 in config.h,did i also need to define OLED_DRIVER = SSD1306?is this SH1106 only a part of SSD1306,or it'a a totally new define?

really glad to see your help.

sigprof, 2023-05-18:
There is only a single OLED driver at the moment — `OLED_DRIVER = SSD1306`; this driver supports SSD1306, SH1106 and (in `develop` only at the moment) SH1107 chips.  By default the SSD1306 chip is assumed (except for some presets like 64X128 and 128X128 that are in `develop` now — these presets select SH1107 automatically); if you have a display with the SH1106 chip, you need to add `#define OLED_IC OLED_IC_SH1106`.

Note that a display with SSD1306 can also work with the driver configured for SH1106 (the main difference between those chips is that SH1106 does not support the horizontal addressing mode); however, if you need 90° or 270° rotation, using the SSD1306 code would be much better if you actually have that chip (the 90° rotation code for SH1106 is currently only in `develop`, and is significantly slower than the code for SSD1306 due to the chip constraints).

KalamityStorm, 2023-05-18:
> 目前只有一个 OLED 驱动器—— `OLED_DRIVER = SSD1306`；此驱动程序支持 SSD1306、SH1106 和（`develop`目前仅支持）SH1107 芯片。默认假定SSD1306芯片（除了现在的64X128和128X128等一些预设`develop`——这些预设自动选择SH1107）；如果你有一个带有 SH1106 芯片的显示器，你需要添加`#define OLED_IC OLED_IC_SH1106`.
> 
> 请注意，带有 SSD1306 的显示器也可以使用为 SH1106 配置的驱动程序（这些芯片之间的主要区别是 SH1106 不支持水平寻址模式）；但是，如果你需要90°或270°旋转，如果你真的有那个芯片，使用SSD1306代码会好得多（SH1106的90°旋转代码目前只在，并且比SSD1306的代码慢得多，`develop`因为芯片限制）。

i've completely understand it,really thanks for your answer!~

closed 2023-05-18 by KalamityStorm