qmk/qmk_firmware #20976: [Bug] Enabling both VIA support and Joystick support causes unexpected behavior

tomatosoupcan, 2023-05-17:
### Describe the Bug

When setting both 

`VIA_ENABLE = yes`
and
`JOYSTICK_ENABLE = yes`

in `rules.mk`

VIA works as expected, however, the joystick is detected but not setup properly.

[image]

With VIA disabled, the joystick shows up and functions as expected.

[image]

Confirmed against multiple Windows environments, and have confirmed that this behavior is not specific to only my device.

### Keyboard Used

katana60/rev1

### Operating System

Windows

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.0
Ψ QMK home: /home/soup/git/qmk_firmware
Ψ Detected Linux (WSL, Fedora Remix for WSL).
Ψ Git branch: master
Ψ Repo version: 0.20.8
⚠ Git has unstashed/uncommitted changes.
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 11.1.0
Ψ Found avr-gcc version 11.2.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 6.4
Ψ Found dfu-programmer version 0.7.2
Ψ Found dfu-util version 0.11
Would you like to clone the submodules? [Y/n] n
Ψ Submodule status:
Ψ - lib/chibios: 2021-12-08 11:37:03 +0000 --  (257302333)
Ψ - lib/chibios-contrib: 2022-04-18 12:52:47 +0200 --  (2a6b73ff)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2021-07-31 15:44:44 +1000 --  (19a5d533f)
Ψ - lib/vusb: 2020-08-16 01:07:21 +0100 --  (bdb53e4)
Ψ - lib/printf: 2020-01-01 18:16:48 +0100 --  (d3b9846)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [X] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

drashna, 2023-05-22:
Does this happen if you use `JOYSTICK_SHARED_EP=no` (either in your rules.mk, or from the command line?

sigprof, 2023-05-22:
This looks like some annoying Windows behavior — I was able to reproduce it with Windows 11 22H2 (22621.1702) and two different onekey firmwares:
```
qmk flash -kb handwired/onekey/promicro -km joystick -e VIA_ENABLE=yes
qmk flash -kb handwired/onekey/promicro -km joystick
```
If you flash the VIA-enabled firmware first and connect the device to a machine which had never seen that USB VID:PID pair before, `joy.cpl` is able to show the joystick axes and buttons.  When you flash a non-VIA firmware and connect the device to the same machine, `joy.cpl` still shows axes and buttons correctly.  However, if after doing that you flash the VIA-enabled firmware again, `joy.cpl` shows just an empty window instead of axes and buttons — so just connecting the device with non-VIA firmware once is enough to break things for the VIA-enabled firmware, even if the VIA-enabled firmware was used first.

What is really annoying is that even deleting the device from the Devices page in Settings or manually removing all related entries in the Device Manager does not fix the problem — after reconnecting the device with a VIA-enabled firmware `joy.cpl` still shows an empty window.

What does help is going to the `HKEY_CURRENT_USER\System\CurrentControlSet\Control\MediaProperties\PrivateProperties\DirectInput` registry key and removing the subkey for the VID:PID pair which matches the problematic device (`VID_FEED&PID_6465` for the onekey firmware, should be `VID_7272&PID_0C2C` for `rominronin/katana60/rev1`).  The subkey gets repopulated when you open `joy.cpl` the next time, and then the joystick axes and buttons are shown again, until you break it by connecting a device with the same VID:PID, but a different firmware configuration.  You will lose any calibration data though (apparently they are stored under that subkey).

sigprof, 2023-05-22:
https://github.com/WootingKb/wootility-issues/issues/31 suggested clearing even more registry keys:
```
HKEY_CURRENT_USER\System\CurrentControlSet\Control\MediaProperties\PrivateProperties\DirectInput\VID_xxxx&PID_yyyy
HKEY_CURRENT_USER\System\CurrentControlSet\Control\MediaResources\Joystick\DINPUT.DLL\JoystickSettings\VID_xxxx&PID_yyyy
HKEY_CURRENT_USER\System\CurrentControlSet\Control\MediaResources\Joystick\DINPUT.DLL\CurrentJoystickSettings
HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\MediaProperties\PrivateProperties\Joystick\OEM\VID_xxxx&PID_yyyy
```