qmk/qmk_firmware #20982: [Bug] Compiled firmware for yandrstudio/yr6095 do not work

daviddgm, 2023-05-18:
### Describe the Bug

I tried to compile and update firmware for yandrstudio/yr6095 with bootloader stm32duino, uf2boot and tinyuf2, but none of them work in this board.

### Keyboard Used

yandrstudio/yr6095

### Link to product page (if applicable)

https://www.aliexpress.com/item/1005004689966255.html

### Operating System

Linux

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

For every firmware that do not work need to program bootloader again.

sigprof, 2023-05-18:
The fact that the bootloader declared for `yandrstudio/yr6095` does not match what is installed on real PCBs is known for some time: https://github.com/qmk/qmk_firmware/pull/18278#issuecomment-1460567301.  Some other Y&R boards using STM32F103 MCUs (or clones) are probably also affected with the same problem.

Unfortunately, it seems that the maintainer of those boards is not really interested in getting them properly supported in upstream QMK (they apparently have [their own fork](https://github.com/Oh-My-Mechanical-Keyboard/qmk_firmware/tree/y_and_r_kb_pool) with some core modifications related to their customized bootloader), and I don't actually have that PCB, so I'm not 100% sure what should be the proper bootloader setup for that hardware.

daviddgm, 2023-05-18:
Hi. Please one basic question, I need to use .hex or .uf2 file to update the board?

sigprof, 2023-05-18:
That depends on which bootloader you actually have on your PCB:
- If you have some version of the `uf2boot` bootloader (that appears as an USB drive to the computer), you need the .uf2 file.
- If you have the `stm32duino` bootloader, you need flash a .bin file using `dfu-util` (QMK Toolbox may be able to do that, because it has support for that bootloader).
- A .hex file is not used by any common bootloaders for F103.

Note that you can't simply convert a .bin file which was compiled for the `stm32duino` bootloader to the .uf2 format, because the firmware intended for use with the `uf2boot` bootloader needs to be compiled for a different base address.

bartman, 2024-06-30:
I need to reflash the bootloader onto my yr6095.  I have found the [Flashing STM32Duino Bootloader](https://docs.qmk.fm/isp_flashing_guide#flashing-stm32duino-bootloader).  Unfortunately the pads on the board are not labeled.  Anyone figure out which pin is which?

[image]

fauxpark, 2024-06-30:
Per the datasheet the third pin from the left appears to be connected to PA13 (SWDIO). From this we can reasonably infer the second pin is PA14 (SWCLK). Then the first pin with the thicker traces is probably VCC, and the remaining pin GND.

bartman, 2024-06-30:
Thanks.  Found the data sheet here: [STM32F103](https://www.st.com/content/ccc/resource/technical/document/datasheet/0d/93/e0/d7/77/bf/4c/54/CD00210843.pdf/files/CD00210843.pdf/jcr:content/translations/en.CD00210843.pdf)

with multimeter I see, from left to right:
* ~0V -- guessing GND
* ~0V -- guessing SWCLK
* ~3.3V -- guessing SWDIO
* ~3.3V -- guessing VCC