qmk/qmk_firmware #21010: [Bug] Cirque trackpad does not consider rotation value for scroll and secondary click gestures in relative mode

harshitgoel96, 2023-05-21:
### Describe the Bug

Hi, 

I am currently using cirque trackpad on my custom keyboard with following settings:

`
#define CIRQUE_PINNACLE_ADDR 0x2A
#define CIRQUE_PINNACLE_CURVED_OVERLAY
#define CIRQUE_PINNACLE_ATTENUATION EXTREG__TRACK_ADCCONFIG__ADC_ATTENUATE_1X
#define POINTING_DEVICE_ROTATION_90
#define CIRQUE_PINNACLE_POSITION_MODE CIRQUE_PINNACLE_RELATIVE_MODE 
#define CIRQUE_PINNACLE_TAP_ENABLE
#define CIRQUE_PINNACLE_SECONDARY_TAP_ENABLE
#define POINTING_DEVICE_GESTURES_SCROLL_ENABLE 
`

The expected behaviour of the trackpad for scroll is to initiate from the right of the trackpad, but on the rotated device its on the top (physical right) and same is for secondary click, it should be triggered from the top right , but due to 90 deg roation, its triggered from top left.

### Operating System

windows

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: C:/Users/harsh/qmk_firmware
Ψ Detected Windows 10 (10.0.22621).
Ψ QMK MSYS version: 1.7.2
Ψ Git branch: p36_unibody
Ψ Repo version: 0.20.5
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

drashna, 2023-05-22:
Not sure if it was you that I was discussing this with on discord, or somebody else. 

The good news is, I think I know where the issue is.  The bad news is, this is hardware dependent.  Eg, this is handled by the pinnacle touch controller's firmware, and not QMK specifically.  

That said, it may be worth disabling the rotation defines, and then editing this line:
https://github.com/qmk/qmk_firmware/blob/725c9f3c85cb48a0a9b6ff3d6232588bb7b41aab/drivers/sensors/cirque_pinnacle.c#L242

Set it to:
```c
    uint8_t feedconfig2 = HOSTREG__FEEDCONFIG2__GLIDE_EXTEND_DISABLE | HOSTREG__FEEDCONFIG2__INTELLIMOUSE_MODE | HOSTREG__FEEDCONFIG2__SWAP_XY_RELATIVE;
```

If that doesn't work, then you're SOL, basically.

harshitgoel96, 2023-05-22:
Hi,
Yes, we were discussing on the discord. I will take a look and getback. thanks

 
 

https://github.com/qmk/qmk_firmware/blob/725c9f3c85cb48a0a9b6ff3d6232588bb7b41aab/drivers/sensors/cirque_pinnacle.c#L242

drashna, 2023-05-22:
> Hi,
> Yes, we were discussing on the discord. I will take a look and getback. thanks

Awesome!  And yeah, digging through the code, the only thing QMK Firmware does is read from the report, and dump the "scroll" data to the "wheelCount" value. Same with the button.  

Checking the data sheet, and the list of registers, this was the only thing that I saw that might work for relative mode.

harshitgoel96, 2023-05-22:
Hi, This brings the location of scroll and tap to left side of the trackpad instead of right. are there any other registers that i can look into?

drashna, 2023-05-23:
that's odd.  In theory, the invert options shouldn't work for anything other than absolute mode, per the data sheet.