qmk/qmk_firmware #21011: [Bug] Keyboard not detected after ChibiOS update

DrakeRichards, 2023-05-21:
### Describe the Bug

### Issue description
Keyboard is not detected by Windows when flashed with firmware compiled after commit ec0908754389017d723e2f656827381e93d8bd8a (QMK 0.19.8).

### Steps to reproduce
1. Checkout branch `rockon` from [my fork](https://github.com/DrakeRichards/qmk_firmware.git).
2. Connect a Stemcell microcontroller to the computer.
3. Run `qmk compile -kb fingerpunch/rockon/v2 -km default`
4. Put the Stemcell into bootloader mode by double-tapping the reset button.
5. Drag and drop the compiled .uf2 firmware file into the drive that appears.
6. Windows will copy the firmware to the keyboard. When the keyboard disconnects, it will not reconnect.

### Keyboard info
Model: Rock On v2 by Fingerpunch
Controller: Stemcell (STM32F411CC06)
Features enabled: OLED, cirque, encoders, RGBlight (issue occurs with or without these features enabled)
Repository: https://github.com/DrakeRichards/qmk_firmware/tree/rockon/keyboards/fingerpunch/rockon

### Keyboard Used

fingerpunch/rockon/v2

### Link to product page (if applicable)

https://fingerpunch.xyz/product/rock-on-v2/

### Operating System

Windows 11

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: /home/qmk/qmk_firmware
Ψ Detected Linux (WSL).
Ψ Git branch: rockon
Ψ Repo version: 0.19.8
Ψ - Latest rockon: 2023-04-26 17:22:00 -0500 (214cac1b05) -- Added basic Rock On stemcell configuration
Ψ - Latest upstream/master: 2023-04-21 14:55:54 +0100 (ab4013a69e) -- NK Plus (#20392)
Ψ - Latest upstream/develop: 2023-04-21 12:54:46 -0700 (ff6855ca4c) -- [Keyboard] Fix nk plus ws2812 config (#20524)
Ψ - Common ancestor with upstream/master: 2023-01-12 11:47:36 +0100 (ec09087543) -- Update ChibiOS[-Contrib], SIO driver, configs (#17915)
Ψ - Common ancestor with upstream/develop: 2023-01-12 11:47:36 +0100 (ec09087543) -- Update ChibiOS[-Contrib], SIO driver, configs (#17915)
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.3.1
Ψ Found avr-gcc version 8.3.0
Ψ Found avrdude version 6.4
Ψ Found dfu-programmer version 0.7.2
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-01-03 19:29:26 +0000 --  (0062927e3)
Ψ - lib/chibios-contrib: 2023-01-11 16:42:27 +0100 --  (a224be15)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2022-09-19 18:02:44 +0200 --  (8d56ea3)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go
```
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

### Notes
- This is likely something caused by my Windows installation. @Sadekbaroudi tested this same firmware on his Rock On board and it worked normally.
- After flashing, Windows Event Viewer's System log shows a warning with Event ID 51: `An error was detected on device \Device\Harddisk3\DR17 during a paging operation.`
- I have tried reinstalling my drivers with Zadig several times. I don't know if I'm doing it right though. I open Zadig with the keyboard disconnected, select "WinUSB (v6.1.7600.16385)" from the "Driver" dropdown, make sure that the arrow is green and no device is selected, then click "Install WCID Driver". The install finishes without issue.
- I have also tried installing the drivers with QMK Toolbox. They again install normally. However, even when using working firmware, the keyboard is never detected in QMK Toolbox, so I don't know if these are the right drivers.

### Help wanted
- From what I understand, Windows 11 changed how drivers are signed (or something to that effect). Could this be the cause of my issue? Is there something I need to configure in Windows?
- What else could I be missing that would let me flash the latest firmware to my board?

drashna, 2023-05-22:
Using a Stemcell?  

Try adding `BOARD = BLACKPILL_STM32_F411` to your rules.mk, and see if that helps.

DrakeRichards, 2023-05-22:
I tried adding that to rules.mk but the behavior didn't change. I then tried changing the MCU to "STM32F411" and BOOTLOADER to "tinyuf2", but that didn't work either. Here's the current contents of rules.mk:

```makefile
# MCU name
MCU = STM32F411

# Bootloader selection
#   Teensy       halfkay
#   Pro Micro    caterina
#   Atmel DFU    atmel-dfu
#   LUFA DFU     lufa-dfu
#   QMK DFU      qmk-dfu
#   ATmega32A    bootloadHID
#   ATmega328P   USBasp
BOOTLOADER = atmel-dfu
CONVERT_TO = stemcell
BOARD = BLACKPILL_STM32_F411
PIN_COMPATIBLE=elite_c
DEFERRED_EXEC_ENABLE = yes

# Build Options
#   change yes to no to disable
#
BOOTMAGIC_ENABLE = no       # Virtual DIP switch configuration
EXTRAKEY_ENABLE = yes       # Audio control and System control
CONSOLE_ENABLE = no        # Console for debug
COMMAND_ENABLE = no        # Commands for debug and configuration
# Do not enable SLEEP_LED_ENABLE. it uses the same timer as BACKLIGHT_ENABLE
SLEEP_LED_ENABLE = no       # Breathing sleep LED during USB suspend
# if this doesn't work, see here: https://github.com/tmk/tmk_keyboard/wiki/FAQ#nkro-doesnt-work
NKRO_ENABLE = no            # USB Nkey Rollover
BACKLIGHT_ENABLE = no       # Enable keyboard backlight functionality

# Either do RGBLIGHT_ENABLE or RGB_MATRIX_ENABLE and RGB_MATRIX_DRIVER
RGBLIGHT_ENABLE = yes
RGB_MATRIX_ENABLE = no
RGB_MATRIX_DRIVER = WS2812

# BEGIN PWM driver: uncomment if using STeMcell as it's a better led driver
# WS2812_DRIVER = pwm
# END PWM driver: uncomment if using STeMcell as it's a better led driver

MIDI_ENABLE = no            # MIDI support
UNICODE_ENABLE = no         # Unicode
BLUETOOTH_ENABLE = no       # Enable Bluetooth with the Adafruit EZ-Key HID
AUDIO_ENABLE = no           # Audio output on port C6
FAUXCLICKY_ENABLE = no      # Use buzzer to emulate clicky switches
ENCODER_ENABLE = yes
OLED_ENABLE = no
# EXTRAFLAGS     += -flto     # macros disabled, if you need the extra space

MOUSEKEY_ENABLE := no  # not required, but enabling for mouse button keys
POINTING_DEVICE_ENABLE := no
#POINTING_DEVICE_DRIVER := cirque_pinnacle_i2c
#OPT_DEFS += -DCIRQUE_ENABLE

PIMORONI_TRACKBALL_ENABLE=no
HAPTIC_ENABLE=no
```

Because Windows Event Viewer is showing an error with the storage drive that appears when flashing, I used Zadig to check what driver that's using. It looks like their are 2 devices when it's in bootloader mode: the one ending with USB ID 00 is for the storage drive.

[image]

[image]

I don't see USBSTOR listed in the [list of known bootloaders](https://docs.qmk.fm/#/driver_installation_zadig?id=list-of-known-bootloaders), so I assume I shouldn't reinstall anything here, right?

DrakeRichards, 2023-05-22:
I should also note that I had a similar issue with an Atmega32u4 keyboard recently: after upgrading from Windows 10 to Windows 11, the keyboard wasn't getting detected after I flashed new firmware. Reinstalling the bootloader drivers with Zadig worked in that case, which is why I suspect it's a similar thing with this one.

DeltaWhy, 2023-05-23:
I am having the same issue. I was able to use git bisect to trace back to this PR where it broke: https://github.com/qmk/qmk_firmware/pull/17915

`BOARD = BLACKPILL_STM32_F411` had no effect, but I was able to get it to work by adding either:

```
EEPROM_DRIVER = transient
```
or:
```
EEPROM_DRIVER = wear_leveling
WEAR_LEVELING_DRIVER = legacy
```

With the second one I was able to use `EE_HANDS` so it appears to be functional.

sigprof, 2023-05-23:
> Controller: Stemcell (STM32F411CC06)

If you are using the current `master` branch, you need to add this workaround to `rules.mk`:
```make
EEPROM_DRIVER = wear_leveling
WEAR_LEVELING_DRIVER = legacy
```
For some reason the workaround for a ChibiOS bug which adds those settings automatically was added only for STM32F401 chips, but not for STM32F411, therefore only STM32F411xE chips work normally.

Alternatively, the code from the `develop` branch (scheduled to go to `master` on 2023 May 28) should work without using the workaround (don't forget to run `qmk git-submodule` — the fix is in ChibiOS).

DrakeRichards, 2023-05-24:
@sigprof That worked perfectly, thank you so much! Glad to hear the fix is coming soon too.

closed 2023-05-24 by DrakeRichards