qmk/qmk_firmware #21020: [Bug] Cannot flash hex to keyboard

notmentaloutlaw, 2023-05-22:
### Describe the Bug

I am on gentoo with the musl overlay. I am have successfully compiled firmware for the ferris sweep keyboard. However, even though lsusb says

```
Bus 001 Device 018: ID 03eb:2ff4 Atmel Corp. atmega32u4 DFU bootloader
```

And I have all the correct dependencies installed. It keeps on saying that it is Waiting for USB serial port - reset your controller now.

### Keyboard Used

ferris/sweep

### Link to product page (if applicable)

I built it myself

### Operating System

Gentoo (musl/selinux/hardened profile)

### qmk doctor Output

```text
qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: /home/tohka/projects/keyboard/qmk_firmware
Ψ Detected Linux.
⚠ Missing or outdated udev rules for 'atmel-dfu' boards. Run 'sudo cp /home/tohka/projects/keyboard/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'kiibohd' boards. Run 'sudo cp /home/tohka/projects/keyboard/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'stm32-dfu' boards. Run 'sudo cp /home/tohka/projects/keyboard/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'apm32-dfu' boards. Run 'sudo cp /home/tohka/projects/keyboard/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'gd32v-dfu' boards. Run 'sudo cp /home/tohka/projects/keyboard/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'bootloadhid' boards. Run 'sudo cp /home/tohka/projects/keyboard/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbasploader' boards. Run 'sudo cp /home/tohka/projects/keyboard/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbtinyisp' boards. Run 'sudo cp /home/tohka/projects/keyboard/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'md-boot' boards. Run 'sudo cp /home/tohka/projects/keyboard/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'caterina' boards. Run 'sudo cp /home/tohka/projects/keyboard/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'hid-bootloader' boards. Run 'sudo cp /home/tohka/projects/keyboard/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
Ψ Git branch: master
Ψ Repo version: 0.19.11
Ψ - Latest master: 2023-02-12 15:09:43 -0500 (9a654e5728) -- [Keyboard] Add Blue Team Pad (#19807)
Ψ - Latest upstream/master: 2023-05-21 18:44:06 +0200 (725c9f3c85) -- Use wayback machine for unreachable site (#20987)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2023-02-12 15:09:43 -0500 (9a654e5728) -- [Keyboard] Add Blue Team Pad (#19807)
Ψ - Common ancestor with upstream/develop: None
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12
Ψ Found avr-gcc version 12
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 1.0.0
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2022-09-18 10:01:17 +0000 --  (0e9d558b5)
Ψ - lib/chibios-contrib: 2022-10-03 18:09:41 +0200 --  (bb8356fb)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/pico-sdk: 2022-09-19 18:02:44 +0200 --  (8d56ea3)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Other keyboard-related software installed

N/A

### Additional Context

I am running this inside a virtual environment and to my understanding this shouldn't impact it. I can not install it outside a venv as that will break my system packages. I tried installing it in a virtual machine that had glibc (artix) with the keyboard passed through but for some reason that couldn't flash the hex either.

sigprof, 2023-05-22:
“Waiting for USB serial port” means that your keyboard is configured to expect the `caterina` bootloader (usually used on Pro Micro controllers), but apparently you have the `atmel-dfu` bootloader (usually used on Elite-C).  You may add `BOOTLOADER = atmel-dfu` to the `rules.mk` file in your keymap to change the bootloader type without editing the keyboard files.

Also you should install the udev rules, as `qmk doctor` advises, unless you are using something else than `udev`/`eudev` to manage device permissions.

notmentaloutlaw, 2023-05-22:
Ah I forgot to do that. I will try that, I have actually used dfu-programmer directly to flash the hex that was compiled for catarina and now unfortunately I can't get it to return to bootloader mode.

notmentaloutlaw, 2023-05-22:
Alright I have done that and it looks like it has flashed (i.e keyboard keys work and the name of the keyboard is listed in lsusb) however after doing the qmk flash command I do get an error at the end

```
Bootloader Version: 0x00 (0)
Checking memory from 0x0 to 0x6FFF...  Not blank at 0x1.
Erasing flash...  Success
Checking memory from 0x0 to 0x6FFF...  Empty.
Checking memory from 0x0 to 0x4DFF...  Empty.
0%                            100%  Programming 0x4E00 bytes...
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
0%                            100%  Reading 0x7000 bytes...
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
Validating...  Success
0x4E00 bytes written into 0x7000 bytes memory (69.64%).
gmake[1]: *** [platforms/avr/flash.mk:173: flash] Error 254
Make finished with errors
gmake: *** [Makefile:414: megami/sweep:akane:flash] Error 1
```

sigprof, 2023-05-22:
Apparently there is some bug in `dfu-programmer` that causes it to detect some error near the very end of the flashing process, even though everything actually went fine.  One hypothesis is that on some machines the disconnect from the bootloader is detected fast enough that `dfu-programmer` gets an error reply on the final command to exit the bootloader.  So you need to look for the success message, and ignore the exit code if the message is present.

notmentaloutlaw, 2023-05-22:
Alright thanks for the information. I am closing this since I have achieved my goal.

closed 2023-05-22 by notmentaloutlaw