qmk/qmk_firmware #21041: [Feature Request] Add support for Megalodon / DOIO Dual Layer Knob Macro Pad (kb03-1)

draeron, 2023-05-25:
### Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [X] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

The 16/30 key variant of this is already supported. Please could we add support for the 3 key version? I believe it uses ATmega32U4.

Maybe @DOIO2022 could help?

Product:
https://www.keebmonkey.com/products/megalodon-dual-layer-knob-macro-pad?_pos=3&_sid=66bf791f1&_ss=r
http://doiokb.com/index.php?m=home&c=View&a=index&aid=252&lang=en

Layout:
https://wiki.keebmonkey.com/kb03-01.json

16 key version:
https://github.com/qmk/qmk_firmware/tree/master/keyboards/doio/kb16

30 key version:
https://github.com/qmk/qmk_firmware/tree/master/keyboards/doio/kb30

draeron, 2023-05-25:
Stupid me flashed the kb30 firmware on mine... it still "work" and the keyboard send some events.

With the kb30 firmware / layour I still get these:

Top row:
K00 => Num Lock
K01 => Num /
K02 => Num X

External Knob:
clockwise => PgUP
counter => PgDown

Inner Knob:
_No Events_

Knob Click => Print

zvecr, 2023-05-25:
> Please refrain from asking maintainers to add support for specific keyboards -- it is unlikely they will have hardware available, and will not be able to help.

We no longer accept issues for "add support for keyboard x".

closed 2023-05-25 by zvecr

draeron, 2023-05-25:
then could you point me to some resources on how to add it then? I to willing to contribute if I can.