qmk/qmk_firmware #21112: qmk_firmware/docs  /newbs_getting_started.md

bthode, 2023-06-02:
### Issue Description

docs/newbs_getting_started.md gives the following brew command to install on Apple Silicon.

`arch -x86_64 brew install qmk/qmk/qmk`

This command fails with Homebrew instructing you to use an alternate command

Error: Cannot install under Rosetta 2 in ARM default prefix (/opt/homebrew)!

To rerun under ARM use:
    arch -arm64 brew install ...
To install under x86_64, install Homebrew into /usr/local.

---

Full alternate command given, 

`arch -arm64 brew install qmk/qmk/qmk`

---
System Info:

Mac Studio with an M1 Max
macOS 13.4 (22F66)
Kernel Version: Darwin 22.5.0
Homebrew 4.0.20

fauxpark, 2023-06-02:
You need to install Homebrew under Rosetta first.

closed 2023-06-02 by fauxpark

bthode, 2023-06-02:
I was able to produce a bin file and flash it onto my Planck via the arm64 version of qmk installed via Homebrew. Maybe using the Rosetta version is no longer necessary?

fauxpark, 2023-06-02:
Rosetta is still required because the `qmk` formula depends on `osx-cross/arm-gcc-bin@8` which is a precompiled toolchain directly from ARM. I've been meaning to switch it over to the `arm-none-eabi-gcc@8` formula I added, but since GitHub Actions *still* has no M1 runners, no bottle will be generated on their repo. Homebrew will fall back to compiling it locally on install, as it probably does for avr-gcc for the same reason.

bthode, 2023-06-02:
Thank you for the detailed explanation.