qmk/qmk_firmware #21203: [Bug] Error Compile LVGL on RP2040

mazWaz, 2023-06-10:
### Describe the Bug

i had error compile on raspberry pico (RP2040) when usin LVGL

```
QMK Firmware 0.17.5
Making handwired/onekey/rp2040 with keymap lvgl

Generating: .build/obj_handwired_onekey_rp2040_lvgl/src/info_deps.d                                 [OK]
arm-none-eabi-gcc.exe (GCC) 10.1.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Generating: .build/obj_handwired_onekey_rp2040/src/info_config.h                                    [OK]
Generating: .build/obj_handwired_onekey_rp2040/src/default_keyboard.c                               [OK]
Generating: .build/obj_handwired_onekey_rp2040/src/default_keyboard.h                               [OK]
Compiling: keyboards/handwired/onekey/onekey.c                                                      [OK]
Compiling: .build/obj_handwired_onekey_rp2040/src/default_keyboard.c                                [OK]
Compiling: quantum/keymap_introspection.c                                                           [OK]
Compiling: quantum/quantum.c                                                                        [OK]
Compiling: quantum/bitwise.c                                                                        [OK]
Compiling: quantum/led.c                                                                            [OK]
Compiling: quantum/action.c                                                                         [OK]
Compiling: quantum/action_layer.c                                                                   [OK]
Compiling: quantum/action_tapping.c                                                                 [OK]
Compiling: quantum/action_util.c                                                                    [OK]
Compiling: quantum/eeconfig.c                                                                       [OK]
Compiling: quantum/keyboard.c                                                                       [OK]
Compiling: quantum/keymap_common.c                                                                  [OK]
Compiling: quantum/keycode_config.c                                                                 [OK]
Compiling: quantum/sync_timer.c                                                                     [OK]
Compiling: quantum/logging/debug.c                                                                  [OK]
Compiling: quantum/logging/sendchar.c                                                               [OK]
Compiling: quantum/logging/print.c                                                                  [OK]
Compiling: quantum/bootmagic/magic.c                                                                [OK]
Compiling: quantum/matrix_common.c                                                                  [OK]
Compiling: quantum/matrix.c                                                                         [OK]
Compiling: quantum/debounce/sym_defer_g.c                                                           [OK]
Compiling: quantum/main.c                                                                           [OK]
Compiling: lib/printf/src/printf/printf.c                                                           [OK]
Compiling: quantum/unicode/utf8.c                                                                   [OK]
Compiling: quantum/color.c                                                                          [OK]
Compiling: quantum/painter/qp.c                                                                     [OK]
Compiling: quantum/painter/qp_internal.c                                                            [OK]
Compiling: quantum/painter/qp_stream.c                                                              [OK]
Compiling: quantum/painter/qgf.c                                                                    [OK]
Compiling: quantum/painter/qff.c                                                                    [OK]
Compiling: quantum/painter/qp_draw_core.c                                                           [OK]
Compiling: quantum/painter/qp_draw_codec.c                                                          [OK]
Compiling: quantum/painter/qp_draw_circle.c                                                         [OK]
Compiling: quantum/painter/qp_draw_ellipse.c                                                        [OK]
Compiling: quantum/painter/qp_draw_image.c                                                          [OK]
Compiling: quantum/painter/qp_draw_text.c                                                           [OK]
Compiling: drivers/painter/tft_panel/qp_tft_panel.c                                                 [OK]
Compiling: drivers/painter/gc9a01/qp_gc9a01.c                                                       [OK]
Compiling: quantum/painter/qp_comms.c                                                               [OK]
Compiling: drivers/painter/comms/qp_comms_spi.c                                                     [OK]
Compiling: quantum/painter/lvgl/qp_lvgl.c                                                           [OK]
Compiling: lib/lvgl/src/extra/layouts/flex/lv_flex.c                                                [OK]
Compiling: lib/lvgl/src/extra/layouts/grid/lv_grid.c                                                [OK]
Compiling: lib/lvgl/src/extra/libs/bmp/lv_bmp.c                                                     [OK]
Compiling: lib/lvgl/src/extra/libs/ffmpeg/lv_ffmpeg.c                                               [OK]
Compiling: lib/lvgl/src/extra/libs/freetype/lv_freetype.c                                           [OK]
Compiling: lib/lvgl/src/extra/libs/fsdrv/lv_fs_fatfs.c                                              [OK]
Compiling: lib/lvgl/src/extra/libs/fsdrv/lv_fs_posix.c                                              [OK]
Compiling: lib/lvgl/src/extra/libs/fsdrv/lv_fs_stdio.c                                              [OK]
Compiling: lib/lvgl/src/extra/libs/fsdrv/lv_fs_win32.c                                              [OK]
Compiling: lib/lvgl/src/extra/libs/gif/gifdec.c                                                     [OK]
Compiling: lib/lvgl/src/extra/libs/gif/lv_gif.c                                                     [OK]
[... 72 lines cut ...]
Ψ Git branch: master
Ψ Repo version: 0.17.5
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.1.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.0
Ψ Found dfu-programmer version 0.7.2
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-01-11 16:42:27 +0100 --  (a224be15)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

mazWaz, 2023-06-11:
> > `QMK Firmware 0.17.5`
> 
> Checking out this tag of the repository I don't actually see an `lvgl` keymap, which means you must have manually attempted to create/copy/download something instead of updating the entire codebase.
> 
> Please [install the latest version of MSYS](https://github.com/qmk/qmk_distro_msys/releases/tag/1.7.2), [update your copy of the repository](https://docs.qmk.fm/#/newbs_git_using_your_master_branch?id=updating-your-master-branch), and try again.

I'm in the master branch, but I haven't updated the tags. I think I'm using a newer version of QMK

zzeneg, 2023-07-06:
As far as I remember, installing latest QMK MSYS **Beta** helped with that.
My workaround is to add `OPT_DEFS = -O2` to `rules.mk`.
Alternatively you can install WSL Ubuntu, setup QMK in normal way and upgrade `arm-none-eabi-gcc` to latest using that guide https://lindevs.com/install-arm-gnu-toolchain-on-ubuntu. I actually recommend it since compiling LVGL on WSL is much faster (you'd have to manually copy uf2 though).