qmk/qmk_firmware #21214: [Feature Request] Improve “rolling press” support

goremykin, 2023-06-11:
### Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [X] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

Hi! First of all, thank you very much for amazing open source software!  I really appreciate it.
I faced some issues with dual role space key - LT(2, KC_SPC).
I use "rolling press" approach a lot while typing, so the first thing I did was turn on "HOLD_ON_OTHER_KEY_PRESS" and set pretty short "TAPPING_TERM". It works perfect when I want to print, for exampel, '='.
```
+-----------------------------------+
| +---------------+                 |
| | LT(2, KC_SPC) |                 |
| +---------------+                 |
|       +--------------+            |
|       | KC_Z         |            |
|       +--------------+            |
+-----------------------------------+
```

Sometimes I really like to press space lazy and slow, so I had to enable "RETRO_TAPPING". And it also works as expected.
The problem occurs when I finish typing a word and put a space after it. In this case, the space works as a "hold":
```
+-----------------------------------+
| +--------------+                  |
| | KC_W         |                  |
| +--------------+                  |
|       +---------------+           |
|       | LT(2, KC_SPC) |           |
|       +---------------+           |
+-----------------------------------+
```

When I make the "TAPPING_TERM" longer, the second scenario works much better, but the first one starts recognize like "space + z". I can't find the balance, one scenario always works worse than the other.

Do you maybe have an option how to force "w + LT(2, KC_SPC)" to be recognized as w + space?

filterpaper, 2023-06-13:
> but the first one starts recognize like "space + z".

`HOLD_ON_OTHER_KEY_PRESS` will immediately hold (layer 2) in your case the moment you press `KC_Z` right after `LT(2, KC_SPC)` within `TAPPING_TERM`. Thus increasing `TAPPING_TERM` would make `HOLD_ON_OTHER_KEY_PRESS` much easier to trigger hold-layer-2 with `KC_Z`.

goremykin, 2023-06-25:
@filterpaper, I recorded two videos to show the problem.
On the first video the tapping term is 100. LT(2, KC_SPC) + Z prints '=' as it should, but in case of W + LT(2, KC_SPC) the space is missing.
```
#define TAPPING_TERM 100
#define HOLD_ON_OTHER_KEY_PRESS_PER_KEY
#define RETRO_TAPPING
```
[TAPPING_TERM100.webm](https://github.com/qmk/qmk_firmware/assets/25577658/8686d5b4-e696-4c83-be1f-4e2088a79419)

On the second video the tapping term is 200. W + LT(2, KC_SPC) prints W + space as it should, but LT(2, KC_SPC) + Z prints space + Z instead of '='.
```
#define TAPPING_TERM 100
#define HOLD_ON_OTHER_KEY_PRESS_PER_KEY
#define RETRO_TAPPING
```
[TAPPING_TERM200.webm](https://github.com/qmk/qmk_firmware/assets/25577658/1017e5d9-196c-48dd-87af-09609c88b36d)

filterpaper, 2023-06-25:
> On the second video the tapping term is 200. `W` + `LT(2, KC_SPC)` prints W + space as it should, but `LT(2, KC_SPC)` + `Z` prints space + Z instead of '='.

Here, you enabled `#define HOLD_ON_OTHER_KEY_PRESS_PER_KEY`, but did you also setup the `get_hold_on_other_key_press` function to return `true` for `LT(2, KC_SPC)`? Otherwise, it defaults to `false`.

goremykin, 2023-06-25:
@filterpaper, Oh, for some reason I thought that I should override it for granular control only.
Now it works as expected, thank you very much! :heart:

closed 2023-06-25 by goremykin