qmk/qmk_firmware #21221: [Bug] Rev7 Mod

amoses12, 2023-06-12:
### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

filterpaper, 2023-06-13:
These are assumptions, based on your brief problem statement:
1. Pressed and released `RSFT_T(KC_ENT)` _within_ `TAPPING_TERM` = Enter key sent to host
2. Pressed and hold `RSFT_T(KC_ENT)` _longer_ than `TAPPING_TERM` with `KC_I` = Expect capital letter `I` but lower case `i` was sent to host ?

amoses12, 2023-06-13:
> These are assumptions, based on your brief problem statement:
> 
> 1. Pressed and released `RSFT_T(KC_ENT)` _within_ `TAPPING_TERM` = Enter key sent to host
> 2. Pressed and hold `RSFT_T(KC_ENT)` _longer_ than `TAPPING_TERM` with `KC_I` = Expect capital letter `I` but lower case `i` was sent to host ?

First I want to make sure we're not confusing mod tap with tapdance (I believe TAPPING_TERM is a tapdance thing). But essentially yes. I have to hold the `RSFT/KC_ENT` key for longer than I'd like. If I hit the key I'm trying to capitalize at a normal typing pace, it sends `KC_ENT` followed by the lowercase letter. (ex `KC_ENT` followed by lowercase `KC_I` as opposed to just `KC_I`)

sigprof, 2023-06-13:
You probably need to [configure tap/hold decision](https://docs.qmk.fm/#/tap_hold?id=tap-or-hold-decision-modes) — depending on your typing habits, you may prefer either `PERMISSIVE_HOLD` or `HOLD_ON_OTHER_KEY_PRESS` (both of those modes also have a per key version, which you may want to use, especially if you also have some one shot keys — those keys are also affected by the tap/hold decision settings).

Some old versions of QMK tried to apply a behavior similar to `HOLD_ON_OTHER_KEY_PRESS` to all `MT()` keys (and only `MT()`, not other dual-role keys like `LT()`); however, that code was somewhat buggy (the implementation was completely different from the current implementation of `HOLD_ON_OTHER_KEY_PRESS`), and has been removed; now the configuration for all dual-role keys is done in the same way, without any special cases for `MT()`.

amoses12, 2023-06-13:
> You probably need to [configure tap/hold decision](https://docs.qmk.fm/#/tap_hold?id=tap-or-hold-decision-modes) — depending on your typing habits, you may prefer either `PERMISSIVE_HOLD` or [...]

Ah that did it! I wasn't aware of the different mod tap modes, this is the first keyboard I've programmed that's needed it. Thank you so much! I'll go ahead and close this issue as well.