qmk/qmk_firmware #21336: [Bug]

Syzole, 2023-06-22:
### Describe the Bug

LEDs stay on GMMK PRO after flashing with qmk firmware

### Keyboard Used

GMMK PRO

### Link to product page (if applicable)

https://www.gloriousgaming.com/products/glorious-gmmk-pro-75-barebone-black

### Operating System

Windows 11

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: D:/Other/qmk_library
Ψ Detected Windows 10 (10.0.22621).
Ψ QMK MSYS version: 1.7.2
Ψ Git branch: master
Ψ Repo version: 0.21.2
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest master: 2023-06-18 15:08:34 -0700 (88640e47ef) -- [refactor] correct layout data (#21289)
Ψ - Latest upstream/master: 2023-06-21 21:35:35 -0700 (cc4ac72f85) -- Igloo Configurator Touch-Up (#21303)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2023-06-18 15:08:34 -0700 (88640e47ef) -- [refactor] correct layout data (#21289)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.1.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.0
Ψ Found dfu-programmer version 0.7.2
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-01-11 16:42:27 +0100 --  (a224be15)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

zvecr, 2023-06-22:
<https://docs.qmk.fm/#/feature_rgb_matrix?id=additional-configh-options>

```c
#define RGB_DISABLE_WHEN_USB_SUSPENDED // turn off effects when suspended
```

Other gmmk boards have this set.

Syzole, 2023-06-22:
I would put this line in the config.h file in the gmmk/pro/config.h?

filterpaper, 2023-06-24:
> I would put this line in the config.h file in the gmmk/pro/config.h?

You can place that in your own keymap's `config.h`

Syzole, 2023-06-24:
it turns off the lights when the computer sleeps now, thanks.

PR #21370 (2023-06-24): Turn off RGB when suspended
## Description

Turn off RGB on the GMMK Pro when suspended

## Types of Changes

- [ ] Core
- [ ] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [x] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* resolves #21336

## Checklist

- [x] My code follows the code style of this project: [**C**](https://docs.qmk.fm/#/coding_conventions_c), [**Python**](https://docs.qmk.fm/#/coding_conventions_python)
- [x] I have read the [**PR Checklist** document](https://docs.qmk.fm [... cut]
closed 2023-06-24 by Syzole